autonomousvision/transfuser #1: errors when "./CarlaUE4.sh -world-port=<port> -opengl"

shuttworth, 2021-04-21:
when
`./CarlaUE4.sh -world-port=<port> -opengl`
errors happened
`bash: port: No such file or directory`

I do not know <port> how to replace

ap229997, 2021-04-21:
Replace ```<port>``` with any suitable port number, e.g., you can just use ```--world-port=2000```

shuttworth, 2021-04-21:
thank you for replying so soon.
when `./CarlaUE4.sh -world-port=2000 -opengl`
errors `4.24.3-0+++UE4+Release-4.24 518 0 Disabling core dumps.`
if it means it is neccessary to install the Unreal Enginee?
or I should choose the solution without display

ap229997, 2021-04-21:
`4.24.3-0+++UE4+Release-4.24 518 0 Disabling core dumps.`
This is normal when running CARLA, it is not an error. You should be able to see the Unreal display window.

You can also check `nvidia-smi`, there should be a `CarlaUE4-Linux-Shipping` process which occupies 900-1100MB of GPU memory.

closed 2021-04-23 by ap229997