autonomousvision/transfuser #4: UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment

raozhongyu, 2021-04-24:
Thanks for your great wrok, i meet this question "UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment" can you give me some question

raozhongyu, 2021-04-24:
I have solve this question

closed 2021-04-24 by raozhongyu

yewon-lee, 2021-06-10:
Hi @raozhongyu I am facing the same issue as you've described. Did you figure out how to solve this error?

mmmxy0413, 2023-03-20:
Hello @raozhongyu May you tell us how to solve this problem? I have met the same question with you~

927087638, 2023-09-25:
Hello @raozhongyu May you tell us how to solve this problem? I have met the same question with you~

Fengtao22, 2024-06-15:
> Hello @raozhongyu May you tell us how to solve this problem? I have met the same question with you~

This reply might be late. But for others whom encounter similar issues, try to check whether the error info contains"RuntimeError: trying to create rpc server for traffic manager; but the system failed to create because of bind error". If it is the case, then "you are trying to create the TrafficManager in a port that is busy" (https://github.com/carla-simulator/scenario_runner/issues/722). So the easy solution is to `

> pkill -9 python 

`