autonomousvision/transfuser #5: terminate called after throwing an instance of 'clmdep_msgpack::v1::type_error'   what():  std::bad_cast

raozhongyu, 2021-04-25:
Hi, thanks for your great work. I meet this error when i collect the data, amd how much the data is enough

ap229997, 2021-04-25:
Can you share a screenshot of the error? Does this error occur on the server side or the client side?

All of our models are trained on ~130k frames from 8 towns and 14 weather conditions.

raozhongyu, 2021-04-27:
Thanks for your reply so fast,  this issue is probably related to this other one, which is a memory leak issue at the LB. [https://github.com/carla-simulator/leaderboard/issues/81](url)

ap229997, 2021-04-27:
I remember seeing this error a few times on the CARLA server side but I haven't figured out the cause or a workaround. It's probably related to the GitHub issue you pointed out. At least in my case, this error was not too frequent, so I'd just restart the server and evaluation.

raozhongyu, 2021-04-27:
Thanks for your reply, i will try it

closed 2021-04-27 by raozhongyu

raozhongyu, 2021-04-28:
Hi, I'm sorry to ask you another question. I'd like to change the label and size of topdown, but I don't find where to change it in auto_pilot.py and map_agents, can you give me some suggestions. Thanks a lot.

ap229997, 2021-04-28:
You can change different parameters of the topdown semantic map in [map_agent.py](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/team_code/map_agent.py#L14)

raozhongyu, 2021-04-28:
I only find the traffic signs, how can i change the vehicle and buildings

ap229997, 2021-04-28:
The vehicles and building classes are already present in the topdown semantic map that you get from CARLA. Traffic lights and stop signs need to be drawn separately because they are not visible in the topdown view.

raozhongyu, 2021-04-29:
Thanks a lot. I get it.