autonomousvision/transfuser #7: Failed to build docker for leaderboard

SunHaoOne, 2021-06-15:
Hello,
Thank you for the wonderful code! I have downloaded and successfully test for evaluation. Now I want to learn how to use the leaderboard for online evaluation. The official documents are not are not detailed enough. However, the code in your leaderboard helps me understand the how to use the dockerfile to build a docker. Then I have checked the `make_docker.sh`, it seems that I need to set the enviroment first.
So the enviroment is set as follows:
```
export CARLA_ROOT=/home/shy/Desktop/carla0.9.10.1
export LEADERBOARD_ROOT=/home/shy/Desktop/transfuser/leaderboard
export SCENARIO_RUNNER_ROOT=/home/shy/Desktop/transfuser/scenario_runner
export TEAM_CODE_ROOT=/home/shy/Desktop/transfuser/leaderboard/team_code
```
Then I tried to run `sh make_docker.sh` and it shows as follows:
```
mv: target '.tmp/PythonAPI/carla/dist/carla-leaderboard.egg' is not a directory
cp: cannot copy a directory, '/home/shy/Desktop/transfuser/leaderboard/', into itself, '.tmp/leaderboard'
Sending build context to Docker daemon  160.6MB
Step 1/38 : FROM nvidia/cuda:9.0-cudnn7-devel-ubuntu16.04
9.0-cudnn7-devel-ubuntu16.04: Pulling from nvidia/cuda
4007a89234b4: Pull complete 
c1de0f9cdfc1: Pull complete 
c8ee6ca703b8: Pull complete 
b39e2761d3d4: Pull complete 
6fd4d5fe5247: Pull complete 
a2aa5ab1b28d: Pull complete 
d115189e3588: Pull complete 
c208532c0f67: Pull complete 
822dc7c71a4b: Pull complete 
05d21eb076f4: Pull complete 
b62c22f14976: Pull complete 
Digest: sha256:eb0f7d58b15893c7d3d676d23e9a4abaaaa6c09019f24bce29a448a9b8a261d8
Status: Downloaded newer image for nvidia/cuda:9.0-cudnn7-devel-ubuntu16.04
 ---> 7b3673a81378
......
......
#
# To activate this environment, use
#
#     $ conda activate python37
#
# To deactivate an active environment, use
#
#     $ conda deactivate

Removing intermediate container 680b222f6271
 ---> 23199a322ea1
Step 11/38 : RUN packages='py_trees==0.8.3 shapely six dictor requests ephem tabulate' 	&& pip3 install ${packages}
 ---> Running in 9e4b3fc610a9
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 7, in <module>
    from pip._internal.cli.main import main
  File "/usr/local/lib/python3.5/dist-packages/pip/_internal/cli/main.py", line 58
    sys.stderr.write(f"ERROR: {exc}")
                                   ^
SyntaxError: invalid syntax
Removing intermediate container 9e4b3fc610a9
The command '/bin/sh -c packages='py_trees==0.8.3 shapely six dictor requests ephem tabulate' 	&& pip3 install ${packages}' returned a non-zero code: 1

```
In the `mv`, it seems that the program failed to move the file and renamed it, only move the file here. And finally in the `py_trees`, I donot know how to fix this bug and after changing another computer, the same error comes.

ap229997, 2021-06-15:
For submitting to the leaderboard, the directory structure needs to be slightly different as described [here](https://leaderboard.carla.org/submit). 

Also, can you share the exact contents of the Dockerfile that you are using to build the docker image?

SunHaoOne, 2021-06-15:
> For submitting to the leaderboard, the directory structure needs to be slightly different as described [here](https://leaderboard.carla.org/submit).
> 
> Also, can you share the exact contents of the Dockerfile that you are using to build the docker image?

Thanks for your quick reply. After checking the official website, I found the directory structure is the same as yours. So I just use the `dockerfile.master`, because the `make_docker.sh ` turned to build the docker by `dockerfile.master`. And do I need to modify the code in the `dockerfile.master`? Thanks.

ap229997, 2021-06-15:
I created a [leaderboard_submission](https://github.com/autonomousvision/transfuser/tree/leaderboard_submission) branch which has detailed instructions on building docker image and submitting to the leaderboard. It has been tested with multiple models so this should work properly. Let me know if you encounter any issues.

SunHaoOne, 2021-06-16:
Hello,
Thanks you again for your branch! That is great, and helps me a lot to understand the leaderboard further. And thanks again for your documents in your branch, I will read it with the code carefully. Besides, I have found and fixed the `pip` error in the `official dockerfile.master`. And later I will update the leaderboard results.
```
Collecting pip
  Downloading https://files.pythonhosted.org/packages/b1/44/6e26d5296b83c6aac166e48470d57a00d3ed1f642e89adc4a4e412a01643/pip-21.1.2.tar.gz (1.6MB)
Building wheels for collected packages: pip
  Running setup.py bdist_wheel for pip: started
  Running setup.py bdist_wheel for pip: finished with status 'error'
  Complete output from command /usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-gaunwJ/pip/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" bdist_wheel -d /tmp/tmpH23pfbpip-wheel- --python-tag cp27:
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'project_urls'
    warnings.warn(msg)
  /usr/lib/python2.7/distutils/dist.py:267: UserWarning: Unknown distribution option: 'python_requires'
    warnings.warn(msg)
  usage: -c [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
     or: -c --help [cmd1 cmd2 ...]
     or: -c --help-commands
     or: -c cmd --help
  
  error: invalid command 'bdist_wheel'
  
  ----------------------------------------
  Failed building wheel for pip
  Running setup.py clean for pip
Failed to build pip
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Not uninstalling pip at /usr/lib/python2.7/dist-packages, outside environment /usr
  Running setup.py install for pip: started
    Running setup.py install for pip: finished with status 'done'
Successfully installed pip-21.1.2
Collecting pip
  Downloading https://files.pythonhosted.org/packages/cd/82/04e9aaf603fdbaecb4323b9e723f13c92c245f6ab2902195c53987848c78/pip-21.1.2-py3-none-any.whl (1.5MB)
Installing collected packages: pip
  Found existing installation: pip 8.1.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed pip-21.1.2
```
I found there is some `pip3` version error with `python3.5`, so after checking, it is neccssary to set `pip3 install --upgrade "pip < 21.0" && \`, and later the terminal shows :
```DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.
```
Then I successfully build the image now.

ap229997, 2021-06-16:
Just make sure to test the docker image locally before submitting to the leaderboard. That's really helpful for debugging.

SunHaoOne, 2021-06-16:
> Just make sure to test the docker image locally before submitting to the leaderboard. That's really helpful for debugging.

Thanks for your quick reply. After checking the local docker image, the `args.txt` seems lost, and shows as follows:
```
========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> [Errno 2] No such file or directory: '/workspace/team_code/model_ckpt/transfuser/args.txt'

Traceback (most recent call last):
  File "/workspace/leaderboard/leaderboard/leaderboard_evaluator.py", line 262, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/workspace/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "/workspace/team_code/transfuser_agent.py", line 34, in setup
    args_file = open(os.path.join(path_to_conf_file, 'args.txt'), 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/workspace/team_code/model_ckpt/transfuser/args.txt'

```
And there is no `args.txt` here, and I just touch `args.txt` (There is the `dummy.txt` here, and do I need to copy the pretrained model to `/workspace/team_code/model_ckpt/transfuser` ). And it shows as follows:
```
========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent

Could not set up the required agent:
> Expecting value: line 1 column 1 (char 0)

Traceback (most recent call last):
  File "/workspace/leaderboard/leaderboard/leaderboard_evaluator.py", line 262, in _load_and_run_scenario
    self.agent_instance = getattr(self.module_agent, agent_class_name)(args.agent_config)
  File "/workspace/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 45, in __init__
    self.setup(path_to_conf_file)
  File "/workspace/team_code/transfuser_agent.py", line 35, in setup
    self.args = json.load(args_file)
  File "/opt/conda/envs/python37/lib/python3.7/json/__init__.py", line 296, in load
    parse_constant=parse_constant, object_pairs_hook=object_pairs_hook, **kw)
  File "/opt/conda/envs/python37/lib/python3.7/json/__init__.py", line 348, in loads
    return _default_decoder.decode(s)
  File "/opt/conda/envs/python37/lib/python3.7/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/opt/conda/envs/python37/lib/python3.7/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
> Registering the route statistics
```
Is there anything else I need to add? Thanks!

ap229997, 2021-06-16:
The lines containing`args.txt` were redundant, I removed them. You need to put the pre-trained model checkpoint in `model_ckpt/transfuser`.

SunHaoOne, 2021-06-16:
> The lines containing`args.txt` were redundant, I removed them. You need to put the pre-trained model checkpoint in `model_ckpt/transfuser`.

Thanks again for your advice, now I can successfully evaluate the docker image in my computer,it shows good RouteCompletionTest. And I guess it can also run in the leaderboard online.  Thanks again for your contribution to this branch, and I will read the code more carefully.

closed 2021-07-11 by SunHaoOne

Abyss-J, 2023-02-14:
When I built the docker, the following error occurred.
`mv: target '.tmp/PythonAPI/carla/dist/carla-leaderboard.egg' is not a directory`
 I didn't find the `carla-leaderboard.egg` or `carla-leaderboard-py3x.egg`file in the directory of the CARLA installation package under `Python API/carla/dist`. And could you give any help?

Abyss-J, 2023-02-14:
XD， I see， thanks a lot！