autonomousvision/transfuser #10: How long does it take to generate the data set? And what is the size of the data set. It took me a day to generate data for only one city.

LordHui, 2021-07-13:


ap229997, 2021-07-13:
Our total dataset size is about 150k frames (130k train + 20k val). It took us 2-3 days to generate the data on 8 1080Ti GPUs.

LordHui, 2021-07-13:
Thanks for your quick reply！
I want to ask by the way how much time does it take to train the various networks in your papers.
I want to find a framework to verify my fusion method, but I don’t seem to be able to afford the computational overhead of your work：）

ap229997, 2021-07-13:
Geometric Fusion and TransFuser take ~2 days, Late Fusion takes 1 day, CILRS and AIM complete easily within a day and LBC takes 2-3 days since it involves training both the teacher and student models. All these estimates are using a single 1080Ti GPU. The training time can also depend on where you are loading the data from. In our case, all the data was stored on the SSD of the GPU node so dataloading was pretty fast. If your data storage has heavy I/O then that can also affect the training time.

ap229997, 2021-07-13:
One more point - we used OpenGL to generate all our data. If you use Vulkan then it'll take a lot longer.

LordHui, 2021-07-14:
To be honest, your work is great. Whether it is an introductory tutorial for novices, or as a baseline for other work. 
But the data generation process is really painful for people with only one or two GPUs. It would be best if you can upload a public data set, whether in google drive or dropbox. 
More importantly, the same data set can avoid some data misuse problems:)

ap229997, 2021-07-14:
The main problem is that our dataset size is quite big. I'll try to reduce the data to only contain the essential elements and see if I can release it.

LordHui, 2021-07-14:
That's great. Thank you so much for your generous contribution to the community!

ap229997, 2021-07-16:
We have released the data (check readme for further information). Let me know if there is any issue with downloading or using data.

daydreamer2023, 2021-07-16:
The download is in progress, although the speed is a bit slow.

[image]

Thank you very much for your active efforts.

ap229997, 2021-07-16:
The download script contains 2 datasets so you can also try to download just the 14_weathers_data if it's taking too long.

daydreamer2023, 2021-07-17:
Thanks for your tips
The download is indeed slower. When I have free time, I should try to delve into the code to see if the data can be generated quickly. Because I'm not sure if the speed is slow due to multiple sensor rendering, or because of the ground truth waypoints (from the expert).

daydreamer2023, 2021-07-17:
Could you release a Mini version that contains the data used in your paper?
It’s a bit too much for me to continue to ask, but I still ask you to think about it.
This may take you some time to delete some data in the total package, maybe this can compress the data to less than 50G.

ap229997, 2021-07-19:
I also released a minimal dataset version (63G) which contains only rgb_front, lidar, and measurements data from the 14_weathers_data. This should be sufficient to train all the models in this repo except LBC.

daydreamer2023, 2021-07-23:
Sorry to reply so late.
In order to run through the code, I made the following changes.（Town*_short->Town*_small ）

`Town01_long   Town02_small  Town03_tiny   Town05_long   Town06_small  Town10_small
Town01_small  Town02_tiny   Town04_long   Town05_small  Town06_tiny   Town10_tiny
Town01_tiny   Town03_long   Town04_small  Town05_tiny   Town07_small
Town02_long   Town03_small  Town04_tiny   Town06_long   Town07_tiny`

https://github.com/autonomousvision/transfuser/blob/963fe48ca8b9b7eb7896a02f97705a48c038f71c/transfuser/config.py#L13-L17

> 
    for town in train_towns:
        train_data.append(os.path.join(root_dir, town+'_tiny'))
        train_data.append(os.path.join(root_dir, town+'_small'))
        if '7' in town or '10' in town:
            continue
        train_data.append(os.path.join(root_dir, town+'_long'))
    for town in val_towns:
        val_data.append(os.path.join(root_dir, town+'_small'))

Finally, thank you again for your contribution.：）

elegentbamboo, 2021-09-08:
> Our total dataset size is about 150k frames (130k train + 20k val). It took us 2-3 days to generate the data on 8 1080Ti GPUs.

I'd like to know how to generate the data on multi-GPUs in parallel, like device'0' for Town01, device'1' for Town02......?Could you describe the generating process in details?

ap229997, 2021-09-08:
We simply ran data generation for each town separately on each GPU, i.e., we ran 8 different data generation scripts, 1 on each GPU. If you want to have a single script, you can use multiprocessing similar to [this](https://github.com/carla-simulator/data-collector/blob/master/multi_gpu_collection.py).

closed 2022-01-24 by ap229997