autonomousvision/transfuser #12: Hello, I want to know what ‘3 training seeds’ mean, can you talk about it in detail?

HongYegg, 2021-08-06:
We report the mean and standard deviation over 9 runs of each method (3 training seeds,
each trained model evaluated 3 times) on Route Completion (RC) and Driving Score (DS) in Town05 Short and Town05
Long settings with scenario and without scenarios. LF: Late Fusion, GF: Geometric Fusion, TF: TransFuser

Hello, I want to know what ‘3 training seeds’ mean, can you talk about it in detail?

HongYegg, 2021-08-06:
> We report the mean and standard deviation over 9 runs of each method (3 training seeds,
> each trained model evaluated 3 times) on Route Completion (RC) and Driving Score (DS) in Town05 Short and [...]

In the experimental design part of the code, I only found the --trafficManagerSeed.

    parser.add_argument('--trafficManagerSeed', default='0',
                        help='Seed used by the TrafficManager (default: 0)')

ap229997, 2021-08-06:
Random seed can affect the training in multiple ways:
- The network parameters are initialized by sampling from some distribution. So, every time you start training, the network is initialized differently unless you fix the torch seed. 
- The dataloader can have multiple sources of randomness: data shuffling, use of other packages like numpy etc. This results in a different data batch every time you rerun training.  

It has been noticed in multiple robotics tasks that there can be significant variance in the evaluation performance depending on the random seed used during training. So, we ran the training multiple times with random seeds (by default, it is randomized in PyTorch unless you set it explicitly) to make sure that the results are consistent.

If you want a fully deterministic training protocol such that the network is always initialized the same, the sequence of data batch remains the same then you need to fix all the random seeds in your code (and any other package that you might be using):
```python
def seed(seed=42):
    np.random.seed(seed)
    random.seed(seed)
    torch.manual_seed(seed)
    torch.cuda.manual_seed(seed)
    torch.cuda.manual_seed_all(seed)
    torch.backends.cudnn.deterministic = True
    torch.backends.cudnn.benchmark = False
```

jun-ja, 2022-01-03:
Now I understand what '3 training seeds' means, but I'm still wondering about the evaluation seeds.
Did you fix the 'trafficManagerSeed' value as default 0 for 3 times evaluation? or did you change it randomly for every evaluation? 
I ask this question because I found that Town05 RouteScenario_23 gives RC score only **2.65** for almost every evaluation and I also found that this is because of the leading vehicle not moving(which is irrelevant to Transfuser's performance since it cannot overtake front vehicles). So I want to modify the movement of this leading vehicle and I want to know if it's possible by changing the random seed.

[image]
(This bicycle rider is not moving till game time is over!)

Thank you!

ap229997, 2022-01-03:
I'm aware of this issue but I don't know how to resolve it.

We did not set the 'trafficManagerSeed' explicitly anywhere so it must be using the default value. The stochasticity in the evaluation comes from the behavior of ego-vehicle and other vehicles in the scene (eg. in CARLA 0.9.10, other vehicles take random turns at intersections which changes the observations of the ego-vehicle resulting in a different behavior). In the newer CARLA versions, the traffic manager has been improved so it should have less stochasticity.

closed 2022-01-24 by ap229997

Watson52, 2022-05-31:
> The CARLA [leaderboard](https://leaderboard.carla.org/leaderboard/) is based on 0.9.10.1 so we use the same.

Such is the case. One more thing, I found that sometimes the "blocked NPC" will disappear. Is it a kind of mechanism of CARLA traffic manager? Thank you in advance for your reply.

ap229997, 2022-05-31:
That happens only when the agent is part of a spawned scenario. If the agent belongs to the background then it does not disappear.