autonomousvision/transfuser #13: Is the calculation of the route length wrong? I feel unreasonable.

HongYegg, 2021-08-17:
Hello, I have some problems and need to ask you for advice. During the test, I carefully analyzed the result.json file I got, and I found that the length of all the routes were almost the same, but I observed the distance the agent was driving on each route, and I could clearly feel that the distances were not exactly the same. of. The test route I ran was Town05-short, and I checked the sample_result.json file you gave and found that the length of the route calculated in it is also the same, which I don’t think is reasonable. Later, I also analyzed the corresponding code for calculating the length of the route, and found that only the Euclidean distance of the two waypoints in Town05-short was calculated, which was unreasonable. In addition, I feel that the unit of distance is not very reasonable.

routeid（in Town05_short.xml）
  <route id="0" town="Town05">
    <waypoint x="72.750000" y="-215.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
    <waypoint x="22.750000" y="-265.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
  </route>
  <route id="1" town="Town05">
    <waypoint x="-163.149994" y="-78.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="-28.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="2" town="Town05">
    <waypoint x="-163.149994" y="11.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="61.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="3" town="Town05">
    <waypoint x="-229.349991" y="12.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-179.349991" y="62.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="4" town="Town05">
    <waypoint x="-9.150002" y="216.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="40.849998" y="266.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>

the corresponding code for calculating the length of the route

def compute_route_length(config):
    trajectory = config.trajectory
    print('trajectory:', trajectory)

    route_length = 0.0
    previous_location = None
    for location in trajectory:
        print('location:', location)
        if previous_location:
            dist = math.sqrt((location.x-previous_location.x)*(location.x-previous_location.x) +
                             (location.y-previous_location.y)*(location.y-previous_location.y) +
                             (location.z - previous_location.z) * (location.z - previous_location.z))
            route_length += dist
        previous_location = location

    return route_length

The printed output is as follows

trajectory: [<carla.libcarla.Location object at 0x7fae161d2cb0>, <carla.libcarla.Location object at 0x7fae161d2cf0>]
location: Location(x=72.750000, y=-215.149994, z=0.000000)
location: Location(x=22.750000, y=-265.149994, z=0.000000)

trajectory: [<carla.libcarla.Location object at 0x7fae161d2e30>, <carla.libcarla.Location object at 0x7fae161d2e70>]
location: Location(x=-163.149994, y=-78.949997, z=0.000000)
location: Location(x=-113.149994, y=-28.949997, z=0.000000)

trajectory: [<carla.libcarla.Location object at 0x7fae161d2fb0>, <carla.libcarla.Location object at 0x7fae161d4030>]
location: Location(x=-163.149994, y=11.599999, z=0.000000)
location: Location(x=-113.149994, y=61.599998, z=0.000000)

trajectory: [<carla.libcarla.Location object at 0x7fae161d4170>, <carla.libcarla.Location object at 0x7fae161d41b0>]
location: Location(x=-229.349991, y=12.450000, z=0.000000)
location: Location(x=-179.349991, y=62.450001, z=0.000000)

trajectory: [<carla.libcarla.Location object at 0x7fae161d42f0>, <carla.libcarla.Location object at 0x7fae161d4330>]
location: Location(x=-9.150002, y=216.349991, z=0.000000)
location: Location(x=40.849998, y=266.350006, z=0.000000)

The above is my analysis process。

Please help to check if there is a problem with unit conversion in this place.This variable route_length_kms
                for key in global_record.infractions.keys():
                    route_length_kms = max(route_record.scores['score_route'] * route_record.meta['route_length'] / 1000.0, 0.001)
[... 1055 lines cut ...]
    "sensors": [
        "carla_camera",
        "carla_camera",
        "carla_camera",
        "carla_camera",
        "carla_lidar",
        "carla_imu",
        "carla_gnss",
        "carla_speedometer"
    ],
    "values": [
        "59.378",
        "85.886",
        "0.717",
        "0.000",
        "1.023",
        "2.872",
        "0.805",
        "0.283",
        "3.686",
        "0.000",
        "3.236",
        "0.316"
    ]
}

HongYegg, 2021-08-17:
Sorry, I'm not very proficient in using it, and my writing is a bit confusing. This is the content not shown above.

<?xml version='1.0' encoding='UTF-8'?>
<routes>
  <route id="0" town="Town05">
    <waypoint x="72.750000" y="-215.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
    <waypoint x="22.750000" y="-265.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
  </route>
  <route id="1" town="Town05">
    <waypoint x="-163.149994" y="-78.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="-28.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="2" town="Town05">
    <waypoint x="-163.149994" y="11.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="61.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="3" town="Town05">
    <waypoint x="-229.349991" y="12.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-179.349991" y="62.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="4" town="Town05">
    <waypoint x="-9.150002" y="216.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="40.849998" y="266.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
</routes>

HongYegg, 2021-08-17:
`<?xml version='1.0' encoding='UTF-8'?>
<routes>
  <route id="0" town="Town05">
    <waypoint x="72.750000" y="-215.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
    <waypoint x="22.750000" y="-265.149994" z="0.0" pitch="0.0" roll="0.0" yaw="90.000122"/>
  </route>
  <route id="1" town="Town05">
    <waypoint x="-163.149994" y="-78.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="-28.949997" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="2" town="Town05">
    <waypoint x="-163.149994" y="11.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-113.149994" y="61.599999" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="3" town="Town05">
    <waypoint x="-229.349991" y="12.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="-179.349991" y="62.450000" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
  <route id="4" town="Town05">
    <waypoint x="-9.150002" y="216.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
    <waypoint x="40.849998" y="266.349991" z="0.0" pitch="0.0" roll="0.0" yaw="270.000114"/>
  </route>
</routes>
`

HongYegg, 2021-08-17:
I think the calculation of the route length may be the config parameter is wrong, and then the unit conversion is not like this, this is what I got when I checked this place.
https://github.com/carla-simulator/leaderboard/blob/master/leaderboard/utils/statistics_manager.py

                 for key in global_record.infractions.keys():
                     route_length_kms = max(route_record.scores['score_route'] / 100 * route_record.meta['route_length'] / 1000.0, 0.001)
                     if isinstance(global_record.infractions[key], list):
                         global_record.infractions[key] = len(route_record.infractions[key]) / route_length_kms
                     else:
                         global_record.infractions[key] += len(route_record.infractions[key]) / route_length_kms

ap229997, 2021-08-17:
That's a good catch. The leaderboard framework computes the route length as the sum of the Euclidean distance between consecutive waypoints where the waypoints are defined in the routes file.

If you check the [route files](https://github.com/carla-simulator/leaderboard/tree/master/data) provided by the official leaderboard repository, you can see that each route is defined by multiple waypoints, not just the start and end waypoints as in our custom routes. When generating the Town05_short routes at intersections, we sample the start & end waypoints at a fixed distance (50m) along 90-degree road segments. This results in Euclidean distance of 50*sqrt(2)=70.17, which is what you see in the results file. 

To get accurate distance as per [compute_route_length](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/leaderboard/utils/statistics_manager.py#L63) function, the route needs to be defined by multiple waypoints and it must contain all the waypoints where there is a change in the navigational command along the route.

Another way to get the correct route distance is to use the interpolated route instead of the provided route in the routes file. The leaderboard framework interpolates the provided route into a dense sequence of waypoints in [route_scenario.py](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/leaderboard/scenarios/route_scenario.py#L221). Ideally this should be used for computing distance accurately.

HongYegg, 2021-08-17:
Excuse me, if this problem exists, does it mean that the results in your paper may not be very rigorous? In addition, is it a bit flawed when converting the unit of route length to kilometers? As I mentioned above.

HongYegg, 2021-08-17:
Thank you very much for your answers.

ap229997, 2021-08-17:
Another point to note here is that this route length bug affects only the infraction metrics (#infractions/km), not driving score (DS) and route completion (RC). In our paper, we use DS and RC as main metrics which are computed accurately. Also, for comparing infractions, we report the infraction count rather than #infractions/km. Since most of our models (AIM, Late Fusion, Geometric Fusion, TransFuser) have similar route completion overall, the comparison is reasonably fair.

As far as unit conversion is concerned, I don't see any issue with it. score_route lies in [0,100] and route_length is in meters, max(..., 0.001) is included to ensure non-zero division when computing #infractions/km in case the agent doesn't even move (minimum traveled distance is 1m, not 0).

HongYegg, 2021-08-17:
Sorry, I misunderstood it, as you said, it does not affect RC and DS. But score_route lies in [0,100] should it be divided by 100 and converted into a percentage.As follows
             for key in global_record.infractions.keys():
                 route_length_kms = max(route_record.scores['score_route'] / 100 * route_record.meta['route_length'] / 1000.0, 0.001)
                 if isinstance(global_record.infractions[key], list):
                     global_record.infractions[key] = len(route_record.infractions[key]) / route_length_kms
                 else:
                     global_record.infractions[key] += len(route_record.infractions[key]) / route_length_kms

ap229997, 2021-08-17:
score_route/100 converts it into a fraction and multiplying it by total route length gives the actual distance traveled by the agent.

closed 2022-01-24 by ap229997