autonomousvision/transfuser #16: about evaluation issues

jun-ja, 2021-09-08:
Hi, I got some issues during evaluation.

1. I got 'list index out of range' error in 'Loading the world'. How can I solve it?
[image]

2. I have often witnessed that vehicles around the ego vehicle sometimes stop without moving and end without proceeding. It seems to occur mainly in the right turn lane. I think this is not a matter of training, but an auto pilot algorithm problem for around vehicles. How can I solve this problem?

This is an example picture of the vehicle stopping and not moving.
[image]

Thank you.

ap229997, 2021-09-08:
1. This occurs when the spawn position of a scenario is already occupied by another agent or a previously spawned scenario. In case of dense urban settings, this can happen a few times. This is not something you should concern with as long as most of the scenarios are getting spawned and only a few are getting skipped. If you really want to avoid this error altogether then you'll have to set the spawn locations in the json scenarios file such that each unique spawn location has only one possible scenario and none of the spawn locations overlap with spawn locations of vehicles.

2. I have also seen cases where vehicles sometimes just get stuck and don't move but I don't think that happens many times. I don't have a solution for this but you'd need to check the traffic manager in CARLA that handles the behavior of other agents and scenarios.

jun-ja, 2021-09-09:
Thank you for your answer!

closed 2021-09-09 by jun-ja