autonomousvision/transfuser #17: About data generation

v413945333, 2021-09-24:
Hello! I have some problems about the data generation.
My operation process is：
1. Running CARLA Server
[image]
2. ./leaderboard/scripts/run_evaluation.sh
[image]
This seems that data generation is not successful. I would like to ask if the above operation is correct, and what should I do if there is a problem. 
And I understand that some images should be generated in the end, where will they be stored?
I would be very grateful if you could reply.

ap229997, 2021-09-24:
The evaluation statistics indicate that the agent has completed the route which means that the data generation is working fine. The data is stored at the location assigned to `SAVE_PATH` in [leaderboard/scripts/run_evaluation.sh](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/scripts/run_evaluation.sh).

v413945333, 2021-09-24:
Thanks for your quick reply! According to your reply, I have found the generated images. They look different from the downloaded data in structure, but it probably doesn't matter.. Thank you again sincerely!

ap229997, 2021-09-24:
Can you the difference in structure in the downloaded and generated images? Is it just the directory names or anything else too?

elegentbamboo, 2021-09-28:
> Can you tell the difference in structure in the downloaded and generated data? Is it just the directory names or anything else too?
My generated data has only one extra *semantic_lidar* directory compared to the structure of *14_weathers_data* as you described.

ap229997, 2021-09-28:
That's fine. I later added the functionality of semantic lidar in the data generation script.

v413945333, 2021-09-28:
Sorry for the late reply, you are right. It is just the difference in directory names.

ZKL-VOILET, 2021-11-18:
Hello! I have some problems about the data generation.
[image]

ap229997, 2021-11-18:
From the error, it seems like the CARLA server cannot bind to port 2000 since it is already in use. Can you verify if the port is available or not?

ZKL-VOILET, 2021-11-18:
> From the error, it seems like the CARLA server cannot bind to port 2000 since it is already in use. Can you verify if the port is available or not?

Thanks for your reply. I have changed the port to 8000, but it stuck here for half an hour. Dose it work?
[image]

ap229997, 2021-11-18:
This message is normal, the server should be running now. You can try connecting a Carla client to the server and verify.

Kin-Zhang, 2022-01-10:
Hi I found that about the data generation agent, did you find it not so correct also about the npc_agent:
I tried this scenario and find things interesting:

[image]
[image]

Like here the auto_agent and npc_agent will have collision and I notice there are many collision test fail on Town03, did you find that maybe a reason for the agent cannot avoid this kind of case?

It just my personal find out, since I tried to do some things more based on this repo.

ap229997, 2022-01-10:
I did not experiment with the npc_agent so I don't know about its failure cases.

Regarding our autopilot, I am aware that it collides a few times. This is because our autopilot uses several heuristics for collision avoidance which has certain blindspots.

Kin-Zhang, 2022-01-11:
> I did not experiment with the npc_agent so I don't know about its failure cases.
> 
> Regarding our autopilot, I am aware that it collides a few times. This is because our autopilot uses several heuristics for collision avoidance which has certain blindspots.

Thanks for the quick reply! I see.
Thanks again.

Kin-Zhang, 2022-01-17:
> thanks for the quick reply! I see.
> Thanks again.

finally, I got the reason since I carefully see how our expert works and notice it didn't consider the change lane things so, the expert agent in this repo have a poor result score on Town03 which has many change lane action in routes.

closed 2022-01-24 by ap229997