autonomousvision/transfuser #20: Problem when run run_evaluation.sh

wanhaodong98, 2021-10-03:
I change agent = transfuser_ agent and agent_ config=best_ model.pth you provided, but the following error occurred. According to the prompt, the torch version should be too low. I upgraded the torch version but reported another error.

wanhaodong98, 2021-10-03:
I use torch==1.4.0

wanhaodong98, 2021-10-03:
I update torch==1.9.1 then

\> Registering the global statistics
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7fb41cdf0440>
Traceback (most recent call last):
  File "/home/harli/transfuser-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 124, in __del__
    self._cleanup()
  File "/home/harli/transfuser-main/leaderboard/leaderboard/leaderboard_evaluator.py", line 157, in _cleanup
    self._agent_watchdog.stop()
  File "/home/harli/transfuser-main/scenario_runner/srunner/scenariomanager/watchdog.py", line 72, in stop
    self._timer.cancel()
AttributeError: 'NoneType' object has no attribute 'cancel'

ap229997, 2021-10-03:
I had updated the model checkpoint for transfuser earlier, which is trained on a newer version. Thanks for pointing this out, I'll update the dependencies.

Regarding the other error, try changing `CHECKPOINT_ENDPOINT` in [run_evaluation.sh](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/scripts/run_evaluation.sh#L21). If the results file of an earlier finished evaluation of the same name already exists and `RESUME=True`, then this error could occur. Let me know if this works.

wanhaodong98, 2021-10-03:
Thank for your reply, I have solved this problem.

closed 2021-10-03 by wanhaodong98

Kin-Zhang, 2021-12-17:
I also met this error, but directly change the pytorch version seems that the car didn't move. when torch==1.9.0 and run_evaluation.sh in docker.

Kin-Zhang, 2021-12-17:
And I also test on the master branch with modify team_agent to transfuser and then run evaluation which I found the agent will not move on first task with some python warning?
[image]
But the success on the second task which is quite wried.