autonomousvision/transfuser #21: Problem about evaluation

wanhaodong98, 2021-10-04:
When I was running the model you provided, I encountered many times that the autonomous vehicle was stationary. I want to know what caused this. The following is the simulation result. The timeout is caused by the vehicle stopping.
https://drive.google.com/file/d/13SiSpYUFG0jfKtFWhHFKHSj0V9TLaemB/view?usp=sharing

Kin-Zhang, 2021-12-17:
Oh! Finally I found some one also notice the lastest model checkpoint have this problem! I think it's quite serious, escpecially you just run it at your computer, here is a picture from the running:
[image]

ap229997, 2021-12-20:
We are aware of this issue and we currently don't have a good solution for it.

FitashUlHaq, 2022-02-03:
any update on this? :)

Kait0, 2022-06-08:
The controller creeping described in the extended paper is a useful workaround to this problem.
https://arxiv.org/pdf/2205.15997.pdf
If you slightly force the model to creep forward it will usually start driving again, if there is no causal reason to stop.

Hither1, 2022-07-03:
> The controller creeping described in the extended paper is a useful workaround to this problem. https://arxiv.org/pdf/2205.15997.pdf If you slightly force the model to creep forward it will usually start driving again, if there is no causal reason to stop.

Hi Just want to clarify this is not implemented in the current git repo code right?

Kait0, 2022-07-04:
We have released the code last week in the [2022 branch](https://github.com/autonomousvision/transfuser/tree/2022) of the github.
You can find it in team_code_latest/submission_agent.py

closed 2022-07-04 by Kait0