autonomousvision/transfuser #30: Traffic light in the image

WPH-commit, 2021-11-18:
Thank you very much for sharing your great work and data. I generated some data using your script and I noticed that usually the traffic light in the image is very small (sometimes even hard for me to distinguish). Is it the common case or I have some wrong settings?

ap229997, 2021-11-18:
That is usually the case and traffic light detection is a problem that I have noticed in the trained models. If you want to generate images with better visibility of traffic lights, you can try playing around with the camera sensor [attributes](https://carla.readthedocs.io/en/latest/ref_sensors/#rgb-camera).

WPH-commit, 2021-11-18:
Thank you very much! But we can't set those advanced attributes of cameras for the carla leaderboard, right?

ap229997, 2021-11-18:
I don't exactly know which all attributes can be set. You might want to check with the CARLA people for that. There's a CARLA discord which you can join and ask about it there.

WPH-commit, 2021-11-26:
Hi, sorry to bother you again. But do you know what these yellow blocks are in the middle of the roads
[image]
?

ap229997, 2021-11-26:
The yellow blocks just in front of the ego vehicle represent some sort of dirt (consider it as an obstacle) on the road which is rendered as part of scenario 1.

closed 2021-11-26 by WPH-commit