autonomousvision/transfuser #31: About Scenarios in training data

WPH-commit, 2021-11-25:
Hi, you mentioned that you include 7 different Scenarios when trainning, according to my undertanding, the event are triggered by "other_actors" in the scenario files, but it seems that your scenario files for each town does not contain other_actors. Is my understanding correct and could you please clarify this? Thank you in advance!

ap229997, 2021-11-25:
That's a good question. While experimenting with CARLA 0.9.10, I had observed that the events are triggered by background agents, not the agents defined in 'other_actors'. I was also surprised by this and a nice way you could verify this is to spawn a single scenario (without any 'other_actors' but make sure there are background agents) yourself. You should be able to see the scenarios spawning correctly. Another interesting thing to do would be to remove any background agents but add a single agent in 'other_actors'. I suspect that in this case 'other_actors' would be used to spawn the scenario. I had not tested this specific case so you could try to verify this too. I don't know if the behavior of the scenarios has changed in the newer CARLA versions.

WPH-commit, 2021-11-25:
Oh that is surprising since I think the background agents are spawned randomly

ap229997, 2021-11-25:
I think it's not exactly the background agents but the agents are explicitly specified in the definition of the scenarios, e.g. check [follow_leading_vehicle.py](https://github.com/autonomousvision/transfuser/blob/main/scenario_runner/srunner/scenarios/follow_leading_vehicle.py#L240)

WPH-commit, 2021-11-25:
Oh I see. Thank you very much! Congratulations on your good places with transfuser based methods on Carla challenge :)

closed 2021-11-25 by WPH-commit