autonomousvision/transfuser #34: Problem in run evaluation for transfuser agent

hlaahmed, 2021-12-01:
I get this error 
[image]

ap229997, 2021-12-01:
From the error, it seems like you don't have permission to create a directory at the location where you are trying to save the data. You should either change the permissions of the location to get write access or comment out [this line](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/scripts/run_evaluation.sh#L23) to avoid saving visualizations while evaluating.

hlaahmed, 2021-12-02:
okay, i added sudo to the command, but now i have this error 
[image]

hlaahmed, 2021-12-02:
The error was because I needed py-trees version less than 1.x

closed 2021-12-02 by hlaahmed