autonomousvision/transfuser #35: Hi, some questions about the model!

roger-cv, 2021-12-02:
Hi, Thanks for this nice work. Recently, I am trying to modify this excellent model to make it suit for my work. As you see, why the "average" operation is required here, the shape of the tensor changes from 64*64 to 8*8 after the "average" operation.  However, the shape of the tensor should be the 64*64 just like the description" fusion at (B,64,64,64)".
[image]

ap229997, 2021-12-02:
The fusion can also be done at 64x64 resolution but that would be too computationally expensive since a transformer is used (quadratic complexity due to attention), so reduce the size to 8x8 at each resolution of the intermediate feature maps.

roger-cv, 2021-12-06:
> 
Thanks for your quick reply. I guess that the input feature map of the transformer of each layer will be downsampled to 8*8 according to what you mean?

ap229997, 2021-12-06:
that's correct, now there are several variants of transformer which address the quadratic complexity issue of the transformer (eg. [Linformer](https://arxiv.org/pdf/2006.04768.pdf)) so maybe it's possible to use the transformer without downsampling.

roger-cv, 2021-12-08:
> that's correct, now there are several variants of transformer which address the quadratic complexity issue of the transformer (eg. [Linformer](https://arxiv.org/pdf/2006.04768.pdf)) so maybe it's possible to use the transformer without downsampling.

Ok, Another interesting question is that can this fusion fashion based on the transformer be replaced with other transformers, such as swim or PVT. Because I notice that this transformer is developed based on the GPT suited for the NLP area.

ap229997, 2021-12-08:
I agree, architecture design can be improved quite a bit.

roger-cv, 2021-12-09:
Ok, Nice work, Thanks for your reply.

Kin-Zhang, 2021-12-17:
But it may require more resources to train...

> I agree, architecture design can be improved quite a bit.

closed 2022-01-24 by ap229997