autonomousvision/transfuser #38: Data preprocessing problems

wensuinan, 2021-12-21:
Hi，Thanks for your great work!
I was wondering what local coordinates are in the code for Transfuse Data? Also, why is the range of X_bins [-32,33] when the point cloud transforms into BEV?

ap229997, 2021-12-21:
The local coordinate space is centered at the ego-vehicle at the current timestep, so we transform the lidar point cloud and ego-vehicle waypoints from future timesteps to the current timestep.

For transforming the point cloud to BEV space, we descretize the BEV space into grids of 0.125m x 0.125m and consider a range of 32m along both x and y axis, resulting in a BEV resolution of 256x256 pixels.

wensuinan, 2021-12-22:
Thanks for you reply!
I still have a few questions.
1. What are the directions of the X,Y, and Z axes of the lidar coordinate, camera coordinate and local coordinate?
2. In the code, X_bin is [-32,33],Y_bin is [-32,0], and X is not 32m in length.
3. below = lidar[lidar[...,2]<=-2.0]，above = lidar[lidar[...,2]>-2.0].  Is 2 the height of the Z axis? Is -2 the height of the 
    ground plane？

Thanks

ap229997, 2021-12-26:
Hi, sorry for the late reply.

1. The local coordinate frame is the ego-vehicle coordinate frame where the Y-axis is to the front of the ego-vehicle, X-axis is to the right of the ego-vehicle and Z-axis is to the bottom of the vehicle (the +ve and -ve direction of these axes vary between different CARLA versions so you need to verify it again for the newer versions). The local coordinate frame is centered at the base of the ego-vehicle. For each sensor coordinate frame, you can check the position of the sensors relative to the center of ego-vehicle in [transfuser_agent.py](https://github.com/autonomousvision/transfuser/blob/main/leaderboard/team_code/transfuser_agent.py) and axes orientation is the same as the local coordinate frame. The LiDAR point cloud that is provided by the sensor is in world coordinate frame.
2. Thanks for pointing out this bug. It should ideally be [-16,16], I somehow missed it and ended up training the models with this configuration. This results in a skewed LiDAR input along the X-axis.
3. This corresponds to the height of the LiDAR sensor and should be -2.3 instead of -2.0. The value is -ve since the Z-axes is inverted in CARLA 0.9.10.

Does this answer your queries?

wensuinan, 2021-12-27:
Thank you for your reply, which provided a good answer to my question.

wensuinan, 2021-12-27:
Thanks you for your reply, which provided a good answer to my questions.

closed 2022-01-24 by ap229997