autonomousvision/transfuser #40: A data generation problem

Halcyon-Lexie, 2021-12-23:
Hi, when I use the autopilot to start data generation, I get a "failure" as the following picture:
[image]
I know this problem is common, but I worry it will affect the final result of model training since we imitate an imperfect expert. How do you think about this?

Halcyon-Lexie, 2021-12-24:
[image]

Halcyon-Lexie, 2021-12-24:
[image]

Halcyon-Lexie, 2021-12-24:
In addition, I don't why I get some black points in the images. Do you know the cause? 
[image]

ap229997, 2021-12-26:
Hi, sorry for the late reply.

The expert uses collision avoidance heuristics which has certain loopholes resulting in few collisions. Ideally, this should be removed from the training data but I think as long as this type of data is very less compared to the proper driving data, the model should be able to learn proper driving behavior.

These black dots are due to some [rendering issues](https://github.com/carla-simulator/carla/issues/3377) when running CARLA with opengl. You can try running with vulkan to remove these artifacts. For running with vulkan, just replace `-opengl` with `-vulkan` when launching the CARLA server.

closed 2022-01-06 by Halcyon-Lexie