autonomousvision/transfuser #41: About the input sequence and PID controller

oskarnatan, 2021-12-26:
Hi @ap229997 , I really like your amazing works. It inspires me a lot. However, there are 2 things that I want to know.

1. In your paper, it is said that the transfuser takes a sequence of inputs. However, it is not described exactly how many inputs the model takes (neither in the supplementary file). In the config.py on each model, I see that you set the parameter
`seq_len = 1 # input timesteps`
Does it mean that each model just takes 1 input and not a sequence of inputs (that contains several inputs)? Could you please explain to me how you feed the inputs to your model?

2. In your paper, it is said that each model (except CILRS) is predicting 4 waypoints (in BEV x,y coordinate). Then, I check the PID controller on the supplementary file and the codes (<model>_agent.py), it seems that only 2 waypoints (the first and second one) are used to control the steer and throttle. So, what is the purpose of predicting 2 extra waypoints? Is it used to give the GRU more learning experience during the training process?

Thanks in advance.

ap229997, 2021-12-26:
1. `seq_len = 1` means we only consider input from a single timestep. Since the transformer takes a sequence of tokens as input, we convert the feature map that we get from the convolutional encoder into a 1D sequence of tokens. For example, 8x8 feature map is converted into 64 tokens, so these 64 tokens form a sequence.
2. Predicting future entities, in general, helps the model to implicitly learn anticipation/forecasting capabilities which is desirable. We experimented with predicting multiple timesteps into the future and 4 timesteps worked the best. Another use of the 4 waypoints could be to design a better controller, e.g. fitting a curve over the 4 waypoints rather than using just a few waypoints.

oskarnatan, 2021-12-26:
Ok, got it. Thanks!

closed 2021-12-26 by oskarnatan