autonomousvision/transfuser #44: Some questions about hyperparameters

pcajilima, 2022-01-26:
Hi! First of all I love the work you've with the TransFuser project, it's really quite interesting and I've had fun diving into autonomous driving research with CARLA thanks to you and your team, so thank you for that!
Currently I am trying to replicate your results and also tune the hyperparameters of the transfuser model so I have a few questions with respect to that.

1. I see that in all of the `train.py` scripts for all of the models, the `--epochs` parameter is set to a default value of 101 except for CILRS which is set to 201, however in the supplementary material you mention that all of the models were trained for 101 epochs. Was the CILRS model really trained at 201 epochs or why is it the only model set to that default value?

2. In the README for each model you give an example of how to train that model, for example, 
`CUDA_VISIBLE_DEVICES=<gpu_id> python3 train.py --id transfuser --batch_size 56` and
`CUDA_VISIBLE_DEVICES=<gpu_id> python3 train.py --id late_fusion --batch_size 128`
I see that for different models you provide different values for `--batch_size` in those examples, however the default values are all set to 24 if I'm not mistaken. What batch size did you use for producing the results found in your main paper? Did you use the same values for all of the models or, like in the examples provided, use different values for different models?

3. During the training process, the model validates itself every 5 epochs, is this so that it can autostop if there's no improvement after a certain number of epochs or what is the reason for this validation? If this is the case, then there wouldn't be much of a need to tune the epochs parameters, correct?

4. From what I can tell, the main tunable parameters when calling the `train.py` script are `--epochs`, `--lr` (learning rate), and `--batch_size`. Do you have any suggestions/recommendations as to how I can go about trying to tune these? Also, are there any other parameters that I didn't mention that could also be fun to experiment with to try to optimize the transfuser model?

pcajilima, 2022-01-26:
1. After how many epochs does the training terminate if the validation loss has not improved? And in what part of the code do you define this termination value? I can't seem to find it.

2. In the [config.py](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/transfuser/config.py) you mention, there is a [learning rate](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/transfuser/config.py#L35) parameter set to 1e-4. Is this parameter equivalent to the one that we define when running, for example, `CUDA_VISIBLE_DEVICES=<gpu_id> python3 train.py --lr 0.0001` or is it for something else? Also, why does the CILRS model not have a learning rate parameter as a training option, but it does have a [learning rate](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/cilrs/config.py#L28) parameter in its config.py file?

I currently have access to a cluster with a few 40GB GPUs so I think I will try to implement a grid search over some of these transformer architecture parameters to take advantage of these resources. This leads me to 2 more questions.

3. I understand that [wandb](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/requirements.txt#L101) is a good tool that can be used for performing a grid search, tho I have never used it nor have I performed a grid search before. Do you have any insight as to how I could go about using this tool and what files I would have to modify/create to perform a grid search of these transformer parameters on a cluster?

4. I know that the transfuser model uses a GPT encoder in its architecture and that those parameters are found in [config.py](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/transfuser/config.py#L42), however I am still fairly new to the topic of attention and I'm not quite sure how all of those parameters affect the overall architecture. Do you happen to have any articles you can recommend for me to read and get a better understanding of how I could tune these transformer parameters?

Again, thank you so much for your insight, I've gained a lot from reading all of the issues posted here and I know they'll also be of use to others!

ap229997, 2022-01-27:
1. This is currently not implemented in the code. I had tensorboard open in the browser and I would just monitor the validation loss manually. But I agree that it should be implemented in the code itself. In our case, validation was done after every 5 epochs and I terminated the training if the validation loss did not improve for 25 epochs.

2. Both the learning rate parameters are the same, you can define them either as a command-line argument or in the config.py file.

3. I'm not that familiar with wandb so I don't know how to run grid search using wandb.

4. For tuning the transformer architecture, you can start with `n_embd`, `n_layer`, `n_head` parameters. I am not aware of any article which specifically focuses on transformer tuning but I'm certain that you should be able to find some resources online.

pcajilima, 2022-01-27:
Thank you for you quick replies. you've been a great help, I will try out some configurations this week and I'll let you know if I find anything interesting!

pcajilima, 2022-01-29:
Hello again, so I've tried training a few models by varying the `n_layer` and `n_head` parameter values in [config.py](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/transfuser/config.py) with {4,8,16} and from what I can tell they all trained successfully, the only notable difference being that the training time increased as `n_layer` increased.

However, when I tried evaluating these models, only the ones that were trained on the default value of `n_layer=8` are evaluating correctly. When I evaluate the models with a different `n_layer` value it takes about a minute to complete and I get the following error for each scenario that it tries to run. Any idea as to what could be the issue here?

[image]

pcajilima, 2022-01-29:
One more question, the `n_layer` parameter refers to the number of encoder layers within the transformer architecture, correct?

pcajilima, 2022-01-30:
Update: I changed a [line](https://github.com/autonomousvision/transfuser/blob/164e01993dc1c67b81ec2680c49c73c2cbba153c/leaderboard/team_code/transfuser_agent.py#L45) in `leaderboard/team_code/transfuser_agent.py` by adding the `strict=False` argument to the function `load_state_dict(path_file, strict=False)`. It took longer than usual for the evaluation to actually start but as far as I can tell it seems to have fixed the issue without any noticeable error as it appears to be evaluating correctly now.

When testing different values of `n_layer` and `n_head` I had trouble implementing a tool to automatically do a grid search so I did it by hand by making various copies of the transfuser model directory and directly modifying the transformer parameters in each respective `config.py` file. My best guess is that when evaluating the modified transfuser models the original `config.py` file with the original parameters gets called at some point leading to the error shown above. My only concern is that I believe this argument only hides the underlying problem rather than actually fixing it, but as long as it continues evaluating correctly I guess it shouldn't be a problem.

ap229997, 2022-01-30:
The error means that you are loading the wrong checkpoint for the model, just verify that the model and checkpoint are the same.

`n_layer` corresponds to the number of self-attention layers in the transformer.

If you set `strict=False` when loading the checkpoint, then checkpoint loading related errors will not be shown. You should use `strict=True` and make sure that the model and checkpoint are same (verify the parameters in the config file).

pcajilima, 2022-02-06:
Sorry for the late reply, it took a while to do a decent amount of testing. You were in fact right, by setting `strict=False` it appears to have loaded the wrong checkpoint resulting in some incorrect and faulty evaluations. I believe this issue was caused because I tried to simultaneously evaluate various transfuser models trained with different `n_layer` and `n_head` parameters. The quick solution I found was to make copies of the directory and modify each config file to match the parameters with which it was trained, that way I could run various evaluations simultaneously.

The only thing that still puzzles me is why the `n_layer` and `n_head` parameters in the config file need to match the values with which each model was trained in order to evaluate that model correctly since I thought those parameters only affected the training process.

ap229997, 2022-02-06:
These parameters are related to the model architecture. So whenever you are loading a checkpoint (which contains the stored weights of the model), you need to match the configuration.

pcajilima, 2022-02-09:
That makes sense, thank you. So after varying the `n_layer` and `n_head` parameters I've noticed that all resulting transfuser models have a similar trend where there is a large gap between the training loss and validation loss (like the ones seen in the image below). Do you think it's because I trained the models on the 14_weathers dataset while all the evaluations take place on a clear noon weather, or is there another reason? Is there a way to reduce this gap?

[image]

ap229997, 2022-02-09:
Validation loss is generally always higher than training loss since the model will eventually start overfitting to training data and get low training loss. The trend is ok, the main thing you need to check is the performance of the model on the evaluation routes.

pcajilima, 2022-02-09:
Ok, thank you. I have one final question before closing this issue.
From what I understand, the PID controller does not affect the training process, only the evaluation of the model. Therefore, I would only need to tune the PID controller hyper-parameters in the config.py file for running an evaluation, not for training a model, correct?

ap229997, 2022-02-09:
that's right

pcajilima, 2022-02-09:
Thank you, I highly appreciate all the help you've given me. I will now close this issue, keep up the good research!

closed 2022-02-09 by pcajilima