autonomousvision/transfuser #46: Question about the version of the dataset provided for downloading.

GuillermoHan97, 2022-02-16:
Thanks for the awesome work! 

I have a question regarding to the dataset. In Transfuser+ it proposed a new agent named 'SEED' and regenerated a training dataset using 'SEED'. So I'd like to know whether the dataset provided for downloading in this repo is the one generated by Transfuser+'s 'SEED', or it is the one generated by original Transfuser Agent?

Thanks in advance :).

ap229997, 2022-02-16:
The provided dataset is generated by the autopilot agent from the original transfuser repo.

I did not understand what you mean by 'a new agent named SEED'. Can you clarify?

GuillermoHan97, 2022-02-16:
Thanks for your quick reply! 

No warries, I misunderstood before. I assumed that the dataset was generated with the Transfuser+'s autopilot agent but this is actually not the case.

closed 2022-02-17 by GuillermoHan97