autonomousvision/transfuser #48: Question about fps and rgb input

Fan-Yixuan, 2022-02-26:
Thanks for your great work and quality code! I have the following questions.

1. Seems that https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/team_code/transfuser_agent.py#L36 will cause https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/team_code/transfuser_agent.py#L183-L207 to always jump over the first frame no matter self.config.seq_len, this seems no big deal but I want to confirm if this is intentional since this part of the code is not needed at all when self.config.seq_len=1.
2. From what I understand, the leaderboard runs at 20 FPS. I don't understand why we get one lidar each two frames https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/team_code/transfuser_agent.py#L246
3. I don't understand why you specifically set the sampling rate for some sensors instead of using the default system frequency (20 FPS) https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/team_code/transfuser_agent.py#L113-L131
4. I saved the picture taken by the front camera during inference and found that there seem to be black waypoints on the picture (visualization of navigation), I want to ask is this normal? Why are they there and can they be removed?
[image]
Thanks again!

ap229997, 2022-02-27:
1. Yes, this part of code is not needed for `seq_len=1`. It is only required when multiple timesteps are considered for input.

2. The default rotation frequency of LiDAR is set to 10 fps in the official CARLA [leaderboard code](
https://github.com/carla-simulator/leaderboard/blob/8956c4e0c53bfa24e2bd0ccb1a5269ce47770a57/leaderboard/autoagents/agent_wrapper.py#L112).

4. We use the same sampling rate for these sensors as provided in the [starter code](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/team_code/base_agent.py).

5. Which agent file did you use to generate this image?

Fan-Yixuan, 2022-02-28:
Thanks for your reply, for the fourth question, I am using transfuser_agent, by adding 
`Image.fromarray(tick_data['rgb']).save('test.png')` 
after
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/team_code/transfuser_agent.py#L177-L181

ap229997, 2022-02-28:
The black waypoints indicate the dense interpolations of the route that is being followed. It seems that debug mode is maybe activated somewhere which is causing the waypoints to show. In my visualizations, this does not happen. Can you check the value of `DEBUG` in `planner.py`. Also, did you make any other changes while running the code?
https://github.com/autonomousvision/transfuser/blob/2af3e0226a3edc45fd5ac3a4545cc790cf04a73f/leaderboard/team_code/planner.py#L7

The mean and scale values are a hack to convert the GPS coordinates into world coordinates. The mean value is the latitude and longitude reference values present in map files for each town. You can find this using:
```python
def _get_latlon_ref(world):
    """
    Convert from waypoints world coordinates to CARLA GPS coordinates
    :return: tuple with lat and lon coordinates
    """
    xodr = world.get_map().to_opendrive()
    tree = ET.ElementTree(ET.fromstring(xodr))

    # default reference
    lat_ref = 42.0
    lon_ref = 2.0

    for opendrive in tree.iter("OpenDRIVE"):
        for header in opendrive.iter("header"):
            for georef in header.iter("geoReference"):
                if georef.text:
                    str_list = georef.text.split(' ')
                    for item in str_list:
                        if '+lat_0' in item:
                            lat_ref = float(item.split('=')[1])
                        if '+lon_0' in item:
                            lon_ref = float(item.split('=')[1])
    return lat_ref, lon_ref
```
Once you get the mean, you can solve for scale by comparing with the world coordinates obtained from the actual transformation function. The correct function for transforming GPS to world coordinates is:
```python
def _location_to_world(lat_ref, lon_ref, location):
	"""
	Convert a location from GPS coordinates to world coordinates
	lat_ref = 49.0, lon_ref = 8.0 for each town
	"""
	EARTH_RADIUS_EQUA = 6378137.0
	scale = math.cos(lat_ref * math.pi / 180.0)
	lat = location['lat']
	lon = location['lon']

	world_x = (lon - lon_ref) * scale * math.pi * EARTH_RADIUS_EQUA / 180.0
	world_y = scale * EARTH_RADIUS_EQUA * math.log(math.tan((90.0 + lat_ref) * math.pi / 360.0) / math.tan((90.0 + lat) * math.pi / 360.0))
	world_z = location['z']

	return (world_x, world_y, world_z)
```

Fan-Yixuan, 2022-03-03:
sorry for the late reply, for the black waypoints, I figured that it was caused by setting the following mode to 1 https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/scripts/run_evaluation.sh#L16
but for the coordinate transformation, I think if we use north as the positive direction, 
`world_y = scale * EARTH_RADIUS_EQUA * math.log(math.tan((90.0 + lat_ref) * math.pi / 360.0) / math.tan((90.0 + lat) * math.pi / 360.0))`
should be 
`world_y = scale * EARTH_RADIUS_EQUA * math.log(math.tan((90.0 + lat) * math.pi / 360.0) / math.tan((90.0 + lat_ref) * math.pi / 360.0))`
could you please share the basis of the formula you gave, thanks a lot!

ap229997, 2022-03-03:
The official [leaderboard repo](https://github.com/carla-simulator/leaderboard/blob/111a48f9099c08a2f1068ee8aea2ad56ce52ef9d/leaderboard/utils/route_manipulation.py#L20) defines the transformation from world coordinates to GPS and I just inverted this function.

closed 2022-06-18 by Kait0