autonomousvision/transfuser #50: How to realize the attention maps？

Watson52, 2022-03-10:
Hi, thanks for your wonderful work especially the detailed code! But I am confused and have some questions about the attention maps. I only have a general understanding of the mechanism of Tranformer. I want to know how to understand the tokens mapping between images and lidar. Specifically, 
1. In the  supplementary, I think these are the same two scenarios, but the green areas appear in two different places when you query the traffic lights in the same location. How do you explain this phenomenon?
[image]

2. Also, in the circled scene above, does the green area in the lidar map change if the traffic light in the image is red? That is to say, does agent's focus change as the traffic lights change?
3. Finally, I just feel confused and can you explain why the lidar is mapped to this location in a situation like this?
[image]

Thank you in advance for your answer!

Watson52, 2022-03-10:
Thank you for your quick response. But I still don't quite understand about the third point that sides of the intersection are mapped to each. Could you please give me some other examples about the visualizations?

Watson52, 2022-03-11:
I see. I will try to run the code later and ask you when I meet a problem. Thank you! Btw, is there another way to download the dataset? I download it slowly on Amazon or sometimes I can't connect to it.

ap229997, 2022-03-11:
Unfortunately, there is no other way to download the dataset.

Watson52, 2022-03-11:
I see. Thank you :)

closed 2022-03-11 by Watson52