autonomousvision/transfuser #51: About the expert in the result

Watson52, 2022-03-11:
Hi, I want to know that the expert with so high score in the result is because of its privileged information? I guess the expert is the agent from the first stage of the LBC.

ap229997, 2022-03-11:
Yes, expert uses privileged information but it is different than the first stage (teacher model) of LBC. Expert uses A-star planner and collision avoidance heuristics to drive.

Watson52, 2022-03-11:
I see. Well, what if the expert did not have privileged information? Or what do you think of the result with the A-star planner and collision avoidance heuristics?

ap229997, 2022-03-11:
The current expert is still not perfect and suffers some infractions since there are certain loopholes in the heuristics.

Watson52, 2022-03-11:
Thanks! I see.

closed 2022-03-11 by Watson52

Watson52, 2022-03-30:
Hi again.
I try to realize the LBC but I didn't quite get it especially the 'topdown'. As you say that the dataset you provided could not train the LBC because of topdown. Could you please tell me what does it mean? Your repply with be appreciated.