autonomousvision/transfuser #52: Dataset

hlaahmed, 2022-03-13:
Hello
Which dataset did you use in training to get the weights in model-ckpt?
Is it 14-weathers-minimal-dataset ?

ap229997, 2022-03-14:
The transfuser checkpoint is trained on a different dataset than the ones provided in this repository. We are currently unable to share the entire dataset.

hlaahmed, 2022-03-14:
What about late fusion checkpoint?

hlaahmed, 2022-03-17:
Hii
I downloaded clear weather dataset
But when I tried to train my model on it
I got this error
I am sure I replaced all paths with mine 
Is the data corrupted ? 
[image]

ap229997, 2022-03-18:
You need to regenerate the preload file, just change the filename `rg_lidar_diag_pl_` to something else in `data.py` of the model directory that you are training
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/aim/data.py#L33

closed 2022-03-22 by hlaahmed