autonomousvision/transfuser #53: error in evaluation

xingbw, 2022-03-14:
hello, thanks for your fantastic work!
I run your code and come across an error while evaluating the model, as shown below:
*****************************
> Setting up the agent
routes_town05_long_03_14_18_56_00
> Loading the world
Skipping scenario 'Scenario4' due to setup error: list index out of range
*****************************
It seems the error occurred when loading the scenario, but I don't know where goes wrong. Did you meet the error like this, or is it normal in evaluation?

xingbw, 2022-03-14:
And I also meet the error as bellow:
**********************************
> Registering the route statistics
ERROR: failed to destroy actor 353 : unable to destroy actor: not found 
ERROR: failed to destroy actor 408 : unable to destroy actor: not found

ap229997, 2022-03-14:
These generally occur if you are trying to load more scenarios/actors than there are spawn points or if there is a conflict in the spawn location with another scenario/actor. This is fine and does not affect the evaluation.

closed 2022-06-18 by Kait0