autonomousvision/transfuser #54: Questions about evaluation

Watson52, 2022-03-17:
Hi, I want to ask you for help. When I run the `run_evaluation.sh` to evaluate the method, I just stuck in the step 'Loading the world' for a long time. When it finally displayed the frames, the screen is very slow and the script will interrupt when the ego-agent into the bend. The following is the fault code.
[image]

`========= Preparing RouteScenario_16 (repetition 0) =========
> Setting up the agent
routes_town05_long_03_17_20_35_11
> Loading the world
Skipping scenario 'Scenario4' due to setup error: list index out of range
> Running the route
leaderboard/team_code/aim_agent.py:169: UserWarning: The given NumPy array is not writeable, and PyTorch does not support non-writeable tensors. This means you can write to the underlying (supposedly non-writeable) NumPy array using the tensor. You may want to copy the array to protect its data or make it writeable before converting it to a tensor. This type of warning will be suppressed for the rest of this program. (Triggered internally at  /pytorch/torch/csrc/utils/tensor_numpy.cpp:180.)
  rgb = torch.from_numpy(scale_and_crop_image(Image.fromarray(tick_data['rgb']), scale=self.config.scale, crop=self.config.input_resolution)).unsqueeze(0)
/home/watson/anaconda3/envs/carla10/lib/python3.7/site-packages/torch/nn/functional.py:718: UserWarning: Named tensors and all their associated APIs are an experimental feature and subject to change. Please do not use them for anything important until they are released as stable. (Triggered internally at  /pytorch/c10/core/TensorImpl.h:1156.)
  return torch.max_pool2d(input, kernel_size, stride, padding, dilation, ceil_mode)
/home/watson/methods/transfuser_1/leaderboard/scripts/run_evaluation.sh: line 38: 22229 Segmentation fault      (core dumped) python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETITIONS} --track=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --record=${RECORD_PATH} --resume=${RESUME} --port=${PORT} --trafficManagerPort=${TM_PORT}
`
And this is the  `run_evaluation.sh` I changed.
`#!/bin/bash

export CARLA_ROOT=carla/CARLA_0.9.10.1
export CARLA_SERVER=${CARLA_ROOT}/CarlaUE4.sh
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla
export PYTHONPATH=$PYTHONPATH:$CARLA_ROOT/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg
export PYTHONPATH=$PYTHONPATH:leaderboard
export PYTHONPATH=$PYTHONPATH:leaderboard/team_code
export PYTHONPATH=$PYTHONPATH:scenario_runner

export LEADERBOARD_ROOT=/home/watson/methods/transfuser_1/leaderboard
export CHALLENGE_TRACK_CODENAME=SENSORS
export PORT=2000 # same as the carla server port
export TM_PORT=8000 # port for traffic manager, required when spawning multiple servers/clients
export DEBUG_CHALLENGE=0
export REPETITIONS=1 # multiple evaluation runs
export ROUTES=leaderboard/data/evaluation_routes/routes_town05_long.xml
export TEAM_AGENT=leaderboard/team_code/aim_agent.py # agent
export TEAM_CONFIG=/home/watson/methods/transfuser_1/aim/log/aim # model checkpoint, not required for expert
export CHECKPOINT_ENDPOINT=results/aim_result.json # results file
export SCENARIOS=leaderboard/data/scenarios/town05_all_scenarios.json
export SAVE_PATH=data/expert # path for saving episodes while evaluating
export RESUME=True

python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py \
--scenarios=${SCENARIOS}  \
--routes=${ROUTES} \
--repetitions=${REPETITIONS} \
--track=${CHALLENGE_TRACK_CODENAME} \
--checkpoint=${CHECKPOINT_ENDPOINT} \
--agent=${TEAM_AGENT} \
--agent-config=${TEAM_CONFIG} \
--debug=${DEBUG_CHALLENGE} \
--record=${RECORD_PATH} \
--resume=${RESUME} \
--port=${PORT} \
--trafficManagerPort=${TM_PORT}
`

ap229997, 2022-03-17:
`town05_all_scenarios.json` contains a lot of scenarios so loading the scenarios takes a few minutes in general. You can switch to `no_scenarios.json` for faster debugging.

Regarding the segmentation fault, are you running the CARLA server and the evaluation code on CPU or GPU?

ap229997, 2022-03-17:
For running the CARLA server on GPU try `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh --world-port=2000 -opengl`

Watson52, 2022-03-18:
> For running the CARLA server on GPU try `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ./CarlaUE4.sh --world-port=2000 -opengl`

Hi, I try the above command then the CARLA started without display. But the run_evaluation.sh still break about save 30 rgb images and the same error.
`leaderboard/scripts/run_evaluation.sh: line 38:  5193 Segmentation fault      (core dumped) python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETITIONS} --track=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --record=${RECORD_PATH} --resume=${RESUME} --port=${PORT} --trafficManagerPort=${TM_PORT}`
[image]

ap229997, 2022-03-18:
Can you tell the exact commands you used for running the server and client code?

It's a bit weird that the code works fine for 30 frames and then crashes. Segmentation fault is generally associated with memory issues.

Watson52, 2022-03-18:
There are two steps for me. For running the server, `./CarlaUE4.sh --world-port=2000 -opengl` in one terminal. And then run `/bin/bash /home/watson/methods/transfuser_1/leaderboard/scripts/run_evaluation.sh` in another terminal. 
I did not try to run client alone and just changed export variable in the run_evaluation.sh. Did I miss something?

ap229997, 2022-03-18:
what GPU are you using?

Watson52, 2022-03-18:
one Titan XP

ap229997, 2022-03-18:
Can you try monitoring GPU memory usage to verify that the server and client code are loading fine on the GPU? Try `watch nvidia-smi` or `nvidia-smi -l 1` and manually check that when you run the server and client code GPU memory increases gradually.

Watson52, 2022-03-18:
I sure the server is running on the GPU. And I found that the GPU memory usage is bigger when I just run ./CarlaUE4.sh without the port and opengl.

ap229997, 2022-03-18:
If you don't specify opengl then CARLA uses vulkan which leads to more GPU memory usage.

The server should take around 1G memory and the client code should also load on GPU. In my case, I have observed segmentation fault when either the server or client is not running on GPU. Apart from that, I am not sure what the issue is.

Watson52, 2022-03-18:
Thanks for your reply. I also found the server takes around 1G and one python3 process takes around 1.3G.

Watson52, 2022-03-18:
I just try the leaderboard official demo and it runs without error. I will find if I have changed some code in your work. Thanks again!

Watson52, 2022-04-18:
Hi, forgive me for bother you again.
When I am trying to generating the data, I don't know how long should set between collecting 2 images. Or call it the interval at which data is saved. I don't know if it will affect the result of training. Could you give me some suggestion?

ap229997, 2022-04-18:
We collect data at 2 FPS, i.e. every 10<sup>th</sup> frame is saved. Increasing the frequency would result in large dataset size which would increase the training time.

Watson52, 2022-04-18:
Thanks for the quick reply! May I ask what's the interval(secnod) between 2 frames? Does it fixed by the Leaderboard?

closed 2022-04-28 by Watson52

Halcyon-Lexie, 2022-05-18:
> I just try the leaderboard official demo and it runs without error. I will find if I have changed some code in your work. Thanks again!

@Watson52  Did you find any cause or solutions? I meet the same error. Does the leaderboard official demo have any difference from the code that the author releases?

Watson52, 2022-05-18:
> @Watson52 Did you find any cause or solutions? I meet the same error. Does the leaderboard official demo have any difference from the code that the author releases?

Do you mean the 'Segmentation fault'? I did not find the specific reason. May be the environment variable or my repo directory. Since the routing problem I have created few repo folders.  
I just deleted the old folders and redownload the repo. Now I could run and evaluate the Transfuser.