autonomousvision/transfuser #57: A question about the coordinate system of the camera and lidar

wcshen, 2022-03-24:
Because the installation yaw of the lidar is -90°, i transform the lidar coordinate from "right-forward-top" to "forward-left-top" through:
`pcds = np.load(lidar_dir + lidar_files[frame]).reshape([-1, 4])
 temp = pcds[:, 0].copy()
 pcds[:, 0] = pcds[:, 1]
 pcds[:, 1] = -temp`
 and then visualize the lidar BEV and corresponding front camera images(the white dot is current target point): 
[image]
In the lidar BEV,the vehicle is turning left,but is turning right in the front camera? 
The data was generated through the script in this repo.

ap229997, 2022-03-24:
Which version of CARLA are you using and why are you doing `pcds[:, 1] = -temp`?

wcshen, 2022-03-24:
CARLA version:0.9.10
I want to transform the lidar coordinate from "right-forward-top" to "forward-left-top:
[image]

ap229997, 2022-03-24:
Now, I get your point. I remember that the coordinate system in CARLA was a bit weird as some axes were inverted and we also faced issues when transforming coordinates. We just corrected for these inversions in our code (in data.py).
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/transfuser/data.py#L226-L243

I agree this is a bit confusing but you also need to correct for these inversions during visualizations.

Kait0, 2022-04-02:
To add to this, the default coordinate system in CARLA is x front y right (z top) which is different from your drawing.
It's a bit hidden but you can check this in the official docs https://carla.readthedocs.io/en/0.9.10/python_api/#carlarotation
Like you said the LiDAR is rotated by 90°, so the initial point cloud you get from the sensor is in y front, x left, (z top).
(You then also have to be careful with the visualization software as they expect a specific coordinate system when visualizing usually)

closed 2022-06-18 by Kait0