autonomousvision/transfuser #59: Simple question regarding the scenarios

xinshuoweng, 2022-04-04:
Hey Aditya, 

Thanks again for your work! I just have a quick question regarding the scenario files here for Town7/10:
https://github.com/autonomousvision/transfuser/blob/main/leaderboard/data/scenarios/town07_all_scenarios.json
https://github.com/autonomousvision/transfuser/blob/main/leaderboard/data/scenarios/town10_all_scenarios.json

These two files seem to be the same as the Town2 scenario file (even the key in the dictionary is still town2 not changed). Is this intentional? Is the Town2 scenario file supposed to work on town7 and 10?

Thanks!
Xinshuo

ap229997, 2022-04-04:
The dense scenario files in this repo are sampled in a grid centered at the locations from the [reference scenarios file](https://github.com/carla-simulator/leaderboard/blob/master/data/all_towns_traffic_scenarios_public.json). Since the reference file does not contain scenarios for Town07, 10, we do not have new scenarios for these. I probably just forgot to remove these. The Town02 scenarios should technically not work on other towns since the map is different but it is possible that some spawn locations just work since the scenario density is quite high.

xinshuoweng, 2022-04-04:
I see. Thanks for the clarification!

closed 2022-04-04 by xinshuoweng