autonomousvision/transfuser #62: data for training

Watson52, 2022-04-28:
Hi
When I trying to generate the data with different scenerios, I have some questions about the dataset.

1. I found the 14_weathers_minimal_data is divide by routes in differenct Town and one route includes kinds of scenarios. Could I divide the dataset by scenario? I mean that there is just one kind of scenario occur in one route. Does it will affect the result of training?
2. There are 3 types of routes: long/short/tiny. I would like to know how the training result will happen without 'long' routes. Or even just have 'tiny' routes, will it make a different to the result?

Looking forward to your help.

ap229997, 2022-04-28:
1. It should not affect training as long as your data contains sufficient diversity of scenarios & routes.
2. The default setting in the paper uses short & tiny routes for training. If you use just 'tiny' routes, the agent will be very conservative (it'll stop a lot). If you use 'long' routes as well then the data distribution gets very skewed towards 'follow lane' situations and the agent may not be able to learn how to react in non-follow lane situations.

Watson52, 2022-04-28:
> 1. It should not affect training as long as your data contains sufficient diversity of scenarios & routes.
> 2. The default setting in the paper uses short & tiny routes for training. If you use just [...]

Thanks! Your explain help me a lot. 
btw, I want to design complex scenarios and use different/separated expert to generate data. Will I success to obtain an agent with experts' abilities? Or it depends on the specific situation to convergence.

Watson52, 2022-04-28:
Thank you for your patience.

Watson52, 2022-06-07:
Hi, I try to generate data using the AutoPilot and I found the setting of lidar sensor only include the carla.Transform. 

1. Does it defaultly  use `channels=64` corresponding to Velodyne 64? Do I need to set other parameters such as `points_per_second`?
2. Why does the value of `yaw `set to `-90`?
https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/leaderboard/team_code/base_agent.py#L169-L174
Thank you in advance.

ap229997, 2022-06-07:
The default settings are:
https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/leaderboard/leaderboard/autoagents/agent_wrapper.py#L149-L164

If you don't set yaw, you would need to rotate the coordinate axes to align them appropriately with the convention used in our code (see #38).

Watson52, 2022-06-08:
Thanks! It helps me a lot. btw, where is the `AgentWrapper `called?

Watson52, 2022-06-08:
Thank you so much :)

Watson52, 2022-06-08:
Hi, sorry to bother you again. I want to change the location of the lidar both for expert and agent, I just need to change the `z` axis coordinate values such as `-2` in the data processing. Is it right? Since I found that the lidar's blind spots are reduced through fine-tuning.
[image]