autonomousvision/transfuser #67: Repetitive tiny routes

xinshuoweng, 2022-05-01:
Hello again!

I just realized that there are a lot of the same tiny routes (generated by the code) and the data generated in these routes seem nearly the same after visualization, which might waste the disk space? Is there any reason not to filter out those routes, e.g., they are useful because of something?

for example, below are two same tiny routes on Town01

  <route id="0" town="Town01">
    <waypoint x="62.008923" y="330.567841" z="0.000000" pitch="0.0" roll="0.0" yaw="0.030693"/>
    <waypoint x="92.418236" y="317.299683" z="0.000000" pitch="0.0" roll="0.0" yaw="269.991272"/>
  </route>

  <route id="2" town="Town01">
    <waypoint x="62.008923" y="330.567841" z="0.000000" pitch="0.0" roll="0.0" yaw="0.030693"/>
    <waypoint x="92.418236" y="317.299683" z="0.000000" pitch="0.0" roll="0.0" yaw="269.991272"/>
  </route>

Thanks in advance!
Xinshuo

ap229997, 2022-05-01:
To generate these routes, I first sampled start and end waypoints at 90 degrees at each intersection ([link](https://github.com/autonomousvision/transfuser/blob/main/tools/generate_intersection_routes.py)) and then extracted each tiny route by explicitly finding the first waypoints where the navigational command changes ([link](https://github.com/autonomousvision/transfuser/blob/main/tools/sample_junctions.py)). In this way, I got 8 routes at each intersection (4 pairs of waypoints and start & end points can be interchanged).

While going through the code, I noticed some bugs. The 8 routes should correspond to: (x-d,y)->(x,y-d), (x-d,y)->(x,y+d), (x,y-d)->(x-d,y), (x,y+d)->(x-d,y), (x+d,y)->(x,y-d), (x+d,y)->(x,y+d), (x,y-d)->(x+d,y), (x,y+d)->(x+d,y), but I mixed up some of these due to which there are duplicates in the tiny routes.
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/tools/generate_intersection_routes.py#L21-L51

The main reason for including these tiny routes was to improve the distribution of turnings and intersections in the dataset (check Fig. 1(a) of [suppmat](http://www.cvlibs.net/publications/Prakash2021CVPR_supplementary.pdf)) since we noticed a significant performance difference between training on the original leaderboard routes and our routes. Removing some of these routes would change the distribution but I don't know if the resulting difference will have a significant impact on the performance.

xinshuoweng, 2022-05-01:
Yep, definitely understand the point of having tiny routes in general. Was only curious about why not filtering out the "repetitive" routes, which however has been clarified by the bug you mentioned. 

It looks like the current code is not updated to resolve this bug yet, right? Would be great if you can update it. I will try on my side as well!

Thanks again!

ap229997, 2022-05-01:
We are currently working on the codebase to remove more bugs and add more functionalities, we'll update it soon.

xinshuoweng, 2022-05-01:
Sounds great! Looking forward to it!

closed 2022-05-01 by xinshuoweng