autonomousvision/transfuser #72: About data processing in the model

Watson52, 2022-05-17:
Hi, when I follow your excellent work I got some issues and need your help.

1. The Lidar data are processed into a two-channel pseudo-image, I feel confused about this process. Does each grid of the pseudo-image is the number of point clouds on/below? Why separate the point cloud into 2 channel according to the Lidar‘s altitude?
2. Assuming that is `t` timestamp right now, do you process future frames of Lidar for `t` to `t+4`? As I know that the expert's trajectory(4 future waypoints) are used to be the ground true. The code in `data.py` :
```
# future frames
for i in range(self.seq_len, self.seq_len + self.pred_len):
lidar_unprocessed = np.load(seq_lidars[i])
full_lidar.append(lidar_unprocessed)
 ```
3. Similar to above, in the `TransFuserAgent.run_step`, does the `enumerate(self.input_buffer['lidar'])` use for the case of `seq_len > 1`? I am confused about does it have some frames in the `input_buffer['lidar']` during the  inference.
4. A potentially naive question, what does the `i` mean for in the `data.py` ?
```
ego_x = seq_x[i]
ego_y = seq_y[i]
ego_theta = seq_theta[i]
 ```

Looking forward to your help. btw, forgive me don't know how to add code efficiently.

ap229997, 2022-05-17:
1. Our LiDAR processing is based on the [code](https://github.com/OATML/oatomobile/tree/alpha/oatomobile/baselines/torch) of [Filos et al](https://arxiv.org/pdf/2006.14911.pdf). It helps the network distinguish between dynamic agents and the ground plane.

2. We don't use the future LiDAR observations, only the current timestep is used.

3. Our code supports incorporating LiDAR inputs from multiple timesteps but we did not observe any benefits in our experiments so we ended using `seq_len=1`.

4. `i` means the current timestep. You can see that we have a 'for' loop just preceding it which runs upto `seq_len`, so `i` just after that refers to the current timestep.
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/transfuser/data.py#L198-L219

Watson52, 2022-05-18:
Thanks for the reply! It helps me a lot. I still have questions. 

1. As proposed at #38 , should I modify the code to this?
`xbins = np.linspace(-x_meters_max, x_meters_max, 2*x_meters_max*pixels_per_meter+1)`

2.  I think the Transfuser do not use the RoadOption or the 'command', instead, using the target point similar to the 'heatmap' in [challenge](https://github.com/bradyz/2020_CARLA_challenge) to output pred_waypoints. It is right?

Watson52, 2022-05-18:
> Thanks for the reply! It helps me a lot. I still have questions.
> 
> 1. As proposed at [Data preprocessing problems #38](https://github.com/autonomousvision/transfuser/issues/38) , should I modify the code to this?
>    `xbins = np.linspace(-x_meters_max, x_meters_max, 2*x_meters_max*pixels_per_meter+1)`
> 2. I think the Transfuser do not use the RoadOption or the 'command', instead, using the target point similar to the 'heatmap' in [challenge](https://github.com/bradyz/2020_CARLA_challenge) to output pred_waypoints. It is right?

For the 2 I notice the AIM method. Thanks for the code again.

ap229997, 2022-05-18:
1. You can modify the code but you won't be able to use the pretrained checkpoints. You would need to train the models again.

2. That is correct, we do not use navigational commands.

Watson52, 2022-05-18:
Thanks for the reply. But I'm still a little confused about the target point.

1. The targer points as input along with the fused features are sent to the GRU to get predict points. In the case when the target point is quite far away, will the pred_waypoints be show as B? Or it's just sort of like A.
[image]

2. In my impression the target points are deal with a `RoutePlanner` that set to a distance in the future about 50m. But I only find one `RoutePlanner` used to set route in the agent file. Could you tell me where is target point's `RoutePlanner` defined?

ap229997, 2022-05-18:
1. More like A since that's how it is present in the training dataset.

2. Check `${CARLA_ROOT}/PythonAPI/carla` where `CARLA_ROOT` is the path to your CARLA installation. You should be able to find the planner at:
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/leaderboard/utils/route_manipulation.py#L16-L18

The global plan is set at:
https://github.com/autonomousvision/transfuser/blob/720c9b566f8b6fdeec991881e291833b1566af9f/leaderboard/leaderboard/scenarios/route_scenario.py#L221-L229

Watson52, 2022-05-18:
Oh...Sure thing the pred_point tends to like the expert's trajectory...
But for the target point I want to find the define just like the [carla challenge](https://github.com/bradyz/leaderboard/blob/7104247204e0e591d43640d2c98490533bc5bbb8/team_code/base_agent.py) using the down sample waypoint for 50m.
```
self._command_planner = RoutePlanner(7.5, 25.0, 257)
self._command_planner.set_route(self._global_plan, True)
```
Thanks for the patience.

ap229997, 2022-05-18:
Yes. The current route is first interpolated into a dense sequence of waypoints using the global planner and then it is downsampled into a sparse set of waypoints. `RoutePlanner` then samples target points from this sparse set of waypoints.

closed 2022-05-18 by Watson52