autonomousvision/transfuser #74: About the trigger scenarios

Watson52, 2022-06-02:
Hi, I found that the `town0X_all_scenarios.json` files including scenarios for 1/3/4/7/8/9/10. But I notice that for scenario 7/8/9 using the same function as shown below. Could I consider that there only have 5 scenarios (1/3/4/789/10) for training or evaluating?
https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/leaderboard/leaderboard/scenarios/route_scenario.py#L54-L65

ap229997, 2022-06-02:
The [scenario definition](https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/scenario_runner/srunner/scenarios/junction_crossing_route.py) has different parameters which can be changed to get different scenarios.

If you look at the [leaderboard description](https://leaderboard.carla.org/scenarios/), scenarios 7,8,9 involve traffic at intersections and the main difference is the direction of the traffic.

Watson52, 2022-06-02:
> The [scenario definition](https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/scenario_runner/srunner/scenarios/junction_crossing_route.py) has different [...]

Thanks for the quick reply! I will check it carefully. btw, the trigger points file such as `town02_all_scenarios.json` using the same points for each scenarios, right?

ap229997, 2022-06-02:
yes