autonomousvision/transfuser #76: Where is the Auxiliary task

raozhongyu, 2022-06-09:
Thanks for your great work. I  have some problems about the auxiliary task. I don't find the auxiliary loss in your train.py of the transfuser. Could you point how to use the auxiliary task?

kashyap7x, 2022-06-09:
In the [original CVPR '21 paper](https://arxiv.org/abs/2104.09224) we do not use any auxiliary loss, so there is no code for this on the current main branch, The code for the [2022 Arxiv extension](https://arxiv.org/abs/2205.15997) will be released on the ["2022" branch](https://github.com/autonomousvision/transfuser/tree/2022) of this repository. We are still cleaning up the training scripts, they will be released soon (and will include the auxiliary tasks).

raozhongyu, 2022-06-09:
Thanks a lot!

raozhongyu, 2022-06-15:
I'd like to konw how long it will be to release the "2022 branch". Thanks a lot

kashyap7x, 2022-06-15:
We will release the data generation script and pre-trained agents by the end of this month. The full release including the training script and additional tools would be in July. Thanks for your patience.

Kait0, 2022-07-04:
We released the code last week on the 2022 branch.
You can find the auxiliary loss in [model.py](https://github.com/autonomousvision/transfuser/blob/4aa19574fc430f23d110de349024830b93cdcf79/team_code_latest/model.py#L732)

closed 2022-07-04 by Kait0

raozhongyu, 2022-07-05:
Thanks a lot!

raozhongyu, 2022-07-12:
> We released the code last week on the 2022 branch. You can find the auxiliary loss in [model.py](https://github.com/autonomousvision/transfuser/blob/4aa19574fc430f23d110de349024830b93cdcf79/team_code_latest/model.py#L732)

Hi, Ihave some questions about the expert data
1、I notice the expert demonstrator for data collection has changed,  would the dataset be updated
2、I also notice that the author has tried the data collection with coach https://github.com/Kait0/carla-roach. Which one is better? Thanks a lot.

kashyap7x, 2022-07-12:
We will release the data collection code and upload the new dataset this week. We did not collect data with roach, we only used it as a baseline to evaluate our new expert demonstrator. On our Longest6 benchmark, Roach obtains a DS of 55.27, whereas our expert gets a DS of 76.91.