autonomousvision/transfuser #81: I'm confused about pi/2 in data.py

muyuuuu, 2022-07-03:
https://github.com/autonomousvision/transfuser/blob/9d5ebf01f7b5f6e5c02ebe0cba423119c465b388/transfuser/data.py#L228-L252

1. in line 233 used `np.pi/2-seq_theta[i]`
2. in line 250 used `np.pi/2+ego_theta`

what's difference between them?

3. in line 239, why `full_lidar[i][:,1] *= -1`

Could you explain it clearly? thx.

ap229997, 2022-07-03:
see #57

muyuuuu, 2022-07-04:
Sorry, I have seen that, but I really don't know why......

Kait0, 2022-07-04:
1-2. The purpose of this code was I think to realign the LiDAR point clouds from multiple time steps into the same coordinate system. Because the code is using self.seq_len = 1 there are no past time steps to realign.
So like you said `ego_theta` and `seq_theta[i]` are the same and that is correct.
3. I don't think there is a good why. It flips the y coordinate axis. 
Think this arose from some confusion about coordinate systems and then was never fixed. You can find it in multiple places in the code so it should be consistent.

muyuuuu, 2022-07-04:
So....... Is there some math manual or coordinate system introduction let me understand `np.pi/2-seq_theta[i]` and `np.pi/2+ego_theta` ?

Kait0, 2022-07-04:
I am not entirely sure what you are confused about.
np.pi/2 is a short form of 90 * np.pi / 180. The first one is a a rotation by 90 degree and the second part converts from degree to radian. 
So np.pi/2 just represents 90° in radian.
I don't really have a recommendation for books on coordinate systems and their transformations, but it should not be too hard to find one on google.

muyuuuu, 2022-07-04:
> this is just something we came up with based on visualizing the alignment issue.

I got it. Thank you very much.

closed 2022-07-04 by Kait0