autonomousvision/transfuser #83: Training Issue: No such file or directory

heming7, 2022-07-19:
Hello transfuser team

First, thank you for the work! I am now facing a issue when try to train the model with the downloaded 2022 dataset. I set the root_dir parameter in train.py to `'/home/username/Work/transfuser/data/coke_dataset_23_11'` but when I ran the train.py, I got the error that says `No such file or directory: '/home/username/Work/transfuser/data/coke_dataset_23_11/Routes_Scenario3_Town07_curved_Seed1000/Scenario3_Town07_curved_route5_11_23_20_22_45/rgb/lidar'`. It seems like it was finding a lidar folder under rgb folder which should not be existed. Is this something wrong with the data.py?

Kait0, 2022-07-21:
Hi,
I think based on the paths you wrote the root_dir should be set to '/home/username/Work/transfuser/data'
The coke_dataset_23_11 folder is just one part of the dataset containing one particular scenario.

heming7, 2022-07-21:
> 

Thanks. It works.

closed 2022-07-21 by heming7