autonomousvision/transfuser #85: Segmentation fault when trying to generate data

NavoditC, 2022-07-21:
Hi,

The Carla server runs successfully on my GeForce GTX 1080 Ti GPU linux machine. However, when I run the command ./leaderboard/scripts/run_evaluation.sh in another terminal window I get the following error:

./leaderboard/scripts/run_evaluation.sh: line 38:  3756 Segmentation fault      (core dumped) python3 ${LEADERBOARD_ROOT}/leaderboard/leaderboard_evaluator.py --scenarios=${SCENARIOS} --routes=${ROUTES} --repetitions=${REPETITIONS} --track=${CHALLENGE_TRACK_CODENAME} --checkpoint=${CHECKPOINT_ENDPOINT} --agent=${TEAM_AGENT} --agent-config=${TEAM_CONFIG} --debug=${DEBUG_CHALLENGE} --record=${RECORD_PATH} --resume=${RESUME} --port=${PORT} --trafficManagerPort=${TM_PORT}

Any suggestions on how this could be rectified would be appreciated. Thanks!

ap229997, 2022-07-22:
Verify once that the port id which running the server and client are the same.

Also, can you provide more information on which agent are you running? are you using the code from `main` branch or `2022`?

NavoditC, 2022-07-22:
Thank you, for your response. On running the command netstat -ltp on terminal I see that on PID 21468 Carla runs and when I running the client I get a segmentation fault.

(Not all processes could be identified, non-owned process info
 will not be shown, you would have to be root to see it all.)
Active Internet connections (only servers)
Proto Recv-Q Send-Q      Local Address                        Foreign Address         State       PID/Program name    
tcp        0           0             localhost:7950                          0.0.0.0:*                 LISTEN      -                   
tcp        0           0             0.0.0.0:cisco-sccp                     0.0.0.0:*                 LISTEN      21468/CarlaUE4-Linux 
tcp        0           0             0.0.0.0:2001                              0.0.0.0:*                 LISTEN      21468/CarlaUE4-Linux 
tcp        0           0             localhost:5939                           0.0.0.0:*                 LISTEN      -                   
tcp        0           0             localhost:domain                       0.0.0.0:*                 LISTEN      -                   
tcp        0           0             localhost:ipp                              0.0.0.0:*                 LISTEN      -                   
tcp6      0           0            ip6-localhost:ipp                         [::]:*                        LISTEN      -                   

The agent I am using is defined in autopilot.py and I am using the main branch. I wanted to mention that I had cloned this repository and installed Carla in DATA on 'Other Locations' in the computer since the memory in home is almost full (Picture attached below). Could that be causing this issue?
[image]

ap229997, 2022-07-22:
Segmentation fault generally occurs due to illegal memory I/O. Hard to say what's causing the issue in your case. You should make sure you have read/write permissions on the relevant locations. Also, check if the value of `world-port` when running CARLA server is the same as value of `port` in `./leaderboard/scripts/run_evaluation.sh`.

NavoditC, 2022-07-22:
I tried on another computer in the lab and everything works fine. The issue was most probably being caused since I was not using the Linux home directory. Thank you very much for your suggestions.

closed 2022-07-22 by NavoditC