autonomousvision/transfuser #86: about the data generation

knifeven, 2022-08-01:
I tried to make a data generation.

During this, I had some issue as follow,

'DataAgent' object has no attribute '_active_traffic_light'

Could you guide me about this?

Kait0, 2022-08-01:
Hi,
can you please provide the route file and scenario file that you use, so that we can reproduce the crash.

knifeven, 2022-08-02:
> Hi, can you please provide the route file and scenario file that you use, so that we can reproduce the crash.

Thanks for the quick mention.

I used the branch "2022" and the command is "./leaderboard/scripts/datagen.sh"

the scenario and route file is as follow,
export SCENARIOS=${WORK_DIR}/leaderboard/data/training/scenarios/Scenario10/Town10HD_Scenario10.json
export ROUTES=${WORK_DIR}/leaderboard/data/training/routes/Scenario10/Town10HD_Scenario10.xml

I also have another issue.
It occurs after running running one scenario in datagen.sh
-----------------------------------------------------------------
Error during the simulation:
> A sensor took too long to send their data
----------------------------------------------------------------

Could you guide me about this?

Thx in advance!

Kait0, 2022-08-05:
Hm I could not reproduce the issue unfortunately.
Based on your description this sounds like a rare initialization problem.
I added a definition of _active_traffic_light to the setup function this should resolve the problem if it occurs.
Could you test the latest commit and report whether this solves your issue?

knifeven, 2022-08-05:
Hi. Thx for your reply

I already modified it.
 --> added "self._active_traffic_light = None" 

Unfortunately, it occurs the same error about "global_map, _vehicle"

I don't know exactly why it is.

Kait0, 2022-08-05:
Thanks for reporting.
I think the underlying crash comes from a case when the simulator sends the HD-map at a later frame than the first one.
Which I think can sometimes happen (the map gets sent only once).
In that case the autopilot would directly return brake, but the data_agent is still trying to save the data and will crash because some variables are not initialized.

I added that the data_agent waits until the HD-Map has arrived in the latest commit.
Can you check if that fixes your problem?

knifeven, 2022-08-10:
Thx for your reply.

I applied the latest commit.

But the error occurred.

========= Preparing RouteScenario_1 (repetition 0) =========
> Setting up the agent
Town10HD_Scenario10_route1_08_10_09_44_34
> Loading the world
> Running the route
Sparse Waypoints: 4
Dense Waypoints: 93
Watchdog exception - Timeout of 49.0 seconds occured
ALSA lib pcm.c:8306:(snd_pcm_recover) underrun occurred

Error during the simulation:
> A sensor took too long to send their data

Traceback (most recent call last):
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/envs/sensor_interface.py", line 239, in get_data
    sensor_data = self._new_data_buffers.get(True, self._queue_timeout)
  File "/usr/lib/python3.7/queue.py", line 178, in get
    raise Empty
_queue.Empty

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 152, in _tick_scenario
    ego_action = self._agent()
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 84, in __call__
    return self._agent()
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/autoagents/autonomous_agent_local.py", line 104, in __call__
    input_data = self.sensor_interface.get_data()
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/envs/sensor_interface.py", line 245, in get_data
    raise SensorReceivedNoData("A sensor took too long to send their data")
leaderboard.envs.sensor_interface.SensorReceivedNoData: A sensor took too long to send their data

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 356, in _load_and_run_scenario
    self.manager.run_scenario()
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 136, in run_scenario
    self._tick_scenario(timestamp)
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 156, in _tick_scenario
    raise RuntimeError(e)
RuntimeError: A sensor took too long to send their data
> Stopping the route
> Registering the route statistics

Kait0, 2022-08-10:
This looks like it is a communication problem between the leaderboard client and carla on your setup.
Some tips that might help you:

You can try increasing the timeout of the watchdog by adding the timeout flag in datagen.sh:
--timeout 600

I see an ALSA lib warning in your output.
If you machine does not have a sound card, you can start carla with the `-nosound ` option.

Make sure all the 3 ports that carla is using are not occupied by other processes.
You set the CARLA ports with the two carla options:
`-carla-rpc-port=${FREE_WORLD_PORT} -carla-streaming-port=${FREE_STREAMING_PORT}`
-carla-rpc-port needs to match the `--port=${FREE_WORLD_PORT}` option in datagen.py (you need to add it).
Also add `--trafficManagerPort=${FREE_TRAFFIC_PORT}` to datagen.
You can check for free ports on the command line using for example:

``FREE_WORLD_PORT=`comm -23 <(seq 10001 10050 | sort) <(ss -Htan | awk '{print $4}' | cut -d':' -f2 | sort -u) | shuf | head -n 1` ``
``FREE_STREAMING_PORT=`comm -23 <(seq 10051 10100 | sort) <(ss -Htan | awk '{print $4}' | cut -d':' -f2 | sort -u) | shuf | head -n 1` ``
``FREE_TRAFFIC_PORT=`comm -23 <(seq 10101 10150 | sort) <(ss -Htan | awk '{print $4}' | cut -d':' -f2 | sort -u) | shuf | head -n 1` ``

So in summary you can try starting carla with:
`/path/to/CarlaUE4.sh -carla-rpc-port=${FREE_WORLD_PORT} -nosound -carla-streaming-port=${FREE_STREAMING_PORT} -opengl `
and add the other two options to datagen.sh

(If you are running CARLA on a machine without a screen additionally add `SDL_VIDEODRIVER=offscreen SDL_HINT_CUDA_DEVICE=0 ` before the /path/to/CarlaUE4.sh )

If the crash you are reporting only happens rarely you can also try implementing a restart script.
We have something like that on our compute cluster, because even if everything is configured correctly CARLA still crashes/has problems occasionally.

knifeven, 2022-08-11:
Thx for the quick reply.

I tried to do your two guide, 
 1. --timeout 600
 2.  /path/to/CarlaUE4.sh -carla-rpc-port=${FREE_WORLD_PORT} -nosound -carla-streaming-port=${FREE_STREAMING_PORT} -opengl

But also occurred the same error.

But I can use "Data Generation" by one scenario.

I appreciate your help.

Thx.

Kait0, 2022-08-16:
I didn't quite understand if this is still a problem.
If so, it would be helpful if you provide the full error logs and a description of what exactly you did, when and how often the error occurs.

If it is not a problem anymore, because you can solve it with restarting etc., we can close the issue.

closed 2022-08-19 by Kait0

EcustBoy, 2023-04-02:
> Hi~ knifeven, recently I meet the same error as you met, i.e. "RuntimeError: A sensor took too long to send their data" when launch data collecting in multi routes, I wanna know have you solved this problem?  I'll thanks much for you reply~

EcustBoy, 2023-04-02:
@knifeven   Hi~ knifeven, recently I meet the same error as you met, i.e. "RuntimeError: A sensor took too long to send their data" when launch data collecting in multi routes, I wanna know have you solved this problem? I'll thanks much for you reply~