autonomousvision/transfuser #88: Data preparation

rginjapan, 2022-08-02:
I have downloaded the minimal dataset, do I still need to do the data generation for running your project?

ap229997, 2022-08-02:
You should download the dataset in the `2022` branch. You don't need to generate it again.

rginjapan, 2022-08-02:
I cannot use Minimal dataset (63G) direclty? Since the dataset in 2022 branch is too big.

Kait0, 2022-08-02:
Minimal dataset is for the cvpr2021 branch, it does not work with the 2022 branch which uses new labels and sensor configurations.

rginjapan, 2022-08-02:
That's fine, I would like to try cvpr2021 branch firstly. Thanks for your reply. So I do not need to generate it again using Minimal dataset (63G) with cvpr2021 branch.

rginjapan, 2022-08-03:
When I did the evaluation of cvpr2021 branch, there is an error messgae but could be skipped:

```
> Setting up the agent
routes_town05_long_08_03_10_36_14
> Loading the world
Skipping scenario 'Scenario4' due to setup error: list index out of range
> Running the route
```

ap229997, 2022-08-03:
This generally occurs when there is no available spawn location for the scenario. The evaluation should still run fine.

closed 2022-08-03 by rginjapan