autonomousvision/transfuser #91: about error, "self._timer.cancel()"

knifeven, 2022-08-05:
I have another issue.

I am using the branch "2022".

after running  ./leaderboard/scripts/local_evaluation.sh, it occurs as following,

--------------------------------------------------------------------------------------------------------------------------------
> Registering the global statistics
Exception ignored in: <function LeaderboardEvaluator.__del__ at 0x7fb8108c78c0>
Traceback (most recent call last):
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 123, in __del__
    self._cleanup()
  File "/mnt/project/transfuser_2022/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 156, in _cleanup
    self._agent_watchdog.stop()
  File "/mnt/project/transfuser_2022/scenario_runner/srunner/scenariomanager/watchdog.py", line 73, in stop
    self._timer.cancel()
---------------------------------------------------------------------------------------------------------------------------------

Could you guide me about it?

Kait0, 2022-08-05:
I think this is an error that hides another crash. The code crashed before the watchdog started leading to the stop function being called before the start function was called.
I pushed a small bugfix to prevent this.
Since there is likely another crash, can you rerun the latest commit and report the new stacktrace?

knifeven, 2022-08-05:
Thx for your reply.

I applied the last commit.
--->if self._timer is not None:
            self._timer.cancel()
Then, it just finished the running program.

the "2022" and "cvpr2021" branch seems quite different.

For reference, run_evaluation.sh in branch of cvpr2021 goes well

Could you guide me about it more?

kashyap7x, 2022-08-05:
Could you please try running the script with `RESUME=0` here?
https://github.com/autonomousvision/transfuser/blob/cb34dc0f5594f5baf53069813138a16e6937905a/leaderboard/scripts/local_evaluation.sh#L20

If you directly see "Registering the global statistics" in the output log, it means the script is trying to resume the evaluation from the existing results file, but there are no remaining routes to run.

knifeven, 2022-08-05:
Thx for the quick reply.

It worked!!!

closed 2022-08-05 by Kait0