autonomousvision/transfuser #92: about using the trained model.

knifeven, 2022-08-15:
Hello.

I generated training data about 0~23 routes according to your guide and trained.
the lidars files (.npy) is 1452 following by:
"Loading 1452 lidars from 1 folders"

Q1: is the lidar count right? I would like to reproduce your trained model. 
      Could you provide me with your trainset information? 

Anyway after trained, I got model_1 ~ 41 and optimizer_1 ~ 41 pth files.

Q2: I wonder which one I could use when I evaluate

(you already provided the three pth files and one args.txt
how did you select the specific pth files?
ex) model_seed1_39.pth, model seed2_39.pth, model_seed3_38.pth)

Kait0, 2022-08-15:
Hi
"Q1: is the lidar count right?" 
No. The full dataset has around 230.000 LiDARs (samples) generated from the around 3000 routes in this repository.
The datagen.sh file only generates data from one route file with one scenario. There are many more routes files in the  [training routes folder](https://github.com/autonomousvision/transfuser/tree/2022/leaderboard/data/training/routes).
Note that re-generating the dataset is computationally expensive. To get it done in a reasonable time you will need to parallelize the data collection. CARLA is nicely parallelizeable as you can run each route file on it's own GPU.
To generate the dataset we used something around 100-300 rtx2080ti GPUs.
We have scripts to run the code in parallel, but we don't release it since it is specific to our custom compute cluster.
If you really want to regenerate the full dataset you will likely need to write your own scripts for running the parallelization on your particular cluster.

For people who don't have access to this kind of computational resources we have released the full dataset for download.
So yes, you can use the download_data.sh it will download the exact dataset that we have used in all our experiments.

"Q2: I wonder which one I could use when I evaluate"
For the ensembles in our main table we have evaluated the checkpoints 31,33,35,37,39 and 41 on the evaluation routes of [NEAT](https://github.com/autonomousvision/neat), and picked the checkpoint with the highest driving score.

This is compute intensive and we have noticed that epoch 31 generally gives results that is very close to the best checkpoint so we use this one for our ablations. This is the epoch right after reducing the learning rate.

"Q3 : How can I train with the trainset provided via ./download_data.sh in 2022 branch?"
How to train a model with the dataset is described in the main readme [here](https://github.com/autonomousvision/transfuser#training-script)
Let me know if anything is unclear about the description.
It is indeed different from the CVPR2021 training script.

knifeven, 2022-08-15:
Thanks for your quick reply
about Q1, I got it.
about Q2, I will train them after downloading your trainset and test it.
about Q3, I understand

Thanks!

closed 2022-08-15 by Kait0