autonomousvision/transfuser #94: download_data.sh Data address is invalid

BeautyCJ, 2022-08-22:
Data address is invalid

Kait0, 2022-08-22:
We will need more information to help you.
What exactly did you try?
What were the full output and error logs?
download_data.sh downloads 13 folders in which of them did the error occur?
As far as I can see the AWS where you can download the data is up and running.

BeautyCJ, 2022-08-22:
when I run download_data.sh, error as follow:

[image]

when I directly past this url in browser, it can't be found
https://s3.eu-central-1.amazonaws.com/avg-projects/transfuser/data/2022_data/left.zip

Kait0, 2022-08-22:
This seems to be a problem with your internet connection. 
I can read the files from my connection.
Since you seem to use on a baidu cluster, I assume you are working in China.
The data is stored on an European aws cluster as you can see.
And EU AWS seems to be blocked.
[image]

I am not super familiar with Chinese internet, but I have heard you need to use a VirtualPrivateNetwork (VPN) to connect to some servers outside of China.

BeautyCJ, 2022-08-22:
ok, i got it, thanks a lot

BeautyCJ, 2022-08-22:
If I use the introduction here （https://github.com/autonomousvision/transfuser#data-generation） to generate data, it is consistent with the downloaded data, right?  Thank you for your help~

Kait0, 2022-08-22:
Yes but only if you generate data for all the training routes.
The example data_gen.sh file only computes the data for 1 of the route files.
To generate the full dataset, you will need to run the script for all the (~70) route files in the [training routes folder](https://github.com/autonomousvision/transfuser/tree/2022/leaderboard/data/training/routes) with the respective scenario file.
I suggest you parallelize across multiple GPUs because generating the full dataset requires a lot of computation.
See also [issue 92](https://github.com/autonomousvision/transfuser/issues/92)

closed 2022-08-23 by Kait0