autonomousvision/transfuser #95: about your trained model.

knifeven, 2022-08-23:
about the trained model you provided, "model_seed1_39.pth",  "model_seed2_39.pth",  "model_seed3_37.pth"

what's mean the seed?

when I try to test loca_evaluation, must be used all three model files?

Kait0, 2022-08-23:
We trained each model 3 times with different random seeds to build an ensemble.
The seed in the name of the file is just an index for the training run (e.g. seed 1 is the model from training run 1, etc.).
The 39 / 37 denote the epoch used.
If you want to reproduce the numbers in the main table you should use all 3 models, yes.
You just need to place them in the same folder and the code will automatically load all three models and average their predictions.

closed 2022-08-23 by Kait0

knifeven, 2022-08-24:
Thx for your reply.

I have one more question.

What is the train set used to make the model you offered?

is that "left ll lr right rl rr s1 s3 s4 s7 s8 s9 s10" total 13 zip files that I can download?