autonomousvision/transfuser #97: Evaluating LatentTF fails

egeonat, 2022-08-24:
Hello,
I am trying to use your pretrained model for the LatentTF architecture. Unfortunately I believe there is a mismatch in the args.txt file corresponding to the LatentTF models. The args.txt file for LatentTF has a line:
`"backbone": "aim",`
yet aim is not a valid backbone accepted by the model.py file. The pretrained state dict also seems to have many different keys compared to the model initialized from creating a model with a LatentTF backbone, which seems odd. 

As it stands I am unable to use the pretrained models of the LatentTF architecture. Any guidance or clarification would be greatly appreciated.

Kait0, 2022-08-25:
Hi,
when we trained the model the name in the code was still aim.
I changed it for the code release to avoid confusion, but apparently forgot to change the args.txt file.
Can you change the line "backbone": "aim" to "backbone": "latentTF" and try again?
I added new features in the training script since training this model, which is why the args.txt file of a new model will have more arguments than the one of the pretrained file.

Can you provide more details on the problem with the keys?

Kait0, 2022-09-02:
We updated the args.txt file. Please report if you encounter any additional issues.

closed 2022-09-02 by Kait0

egeonat, 2022-09-06:
Hello, thank you for your help. You are right, just changing the backbone in the args.txt fixed the issue, I was able to use the pretrained model successfully. 

About the state_dict keys, the provided state_dict had an additional lidar_encoder key that didn't exist in the LatentTF model, but since LatentTF doesn't use such an encoder anyway, simply loading the state_dict with strict=False arguments worked fine. Thanks a lot for sharing your code and taking the time to assist everyone with their issues, I find this repo immensely helpful.