autonomousvision/transfuser #100: Coordinate transformation

KP-Zhang, 2022-08-31:
Hi,
Thank you for your wanderful work.
I am going to transform the coordinate of lidar data (point cloud) as follows.
x-axis points towards to front direction
y-axis points towards to the left direction
z-axis points towards to the top direction
How can I do that?
Thank you for your attention.

Kait0, 2022-08-31:
Depends on where you read the point cloud.
I assume in the following that you directly take it from the tick_data['lidar'].
The default coordinate system that CARLA uses is x front, y right, z up, see [documentation](https://carla.readthedocs.io/en/0.9.10/python_api/#carlarotation).
However our LiDAR sensor is placed rotated by -90° around the yaw [on the car](https://github.com/autonomousvision/transfuser/blob/228a334b7fd98228e083f7c3cc2e7533d1f94b3e/team_code_transfuser/submission_agent.py#L178)
The reason for that is that the LiDAR operates at 10 Hz while the cameras operate at 20 Hz.
So we only get a half LiDAR at every time step, by default right and left.
We rotate the LiDAR to get front and back instead (and then use only the front frames by doing an action repeat of 2 for the model).

So the LiDAR point cloud will be in x left, y front, z up with respect to the car.

To get your coordinate system you would need to rotate the data by 90° and then flip the y axis.
Something like this:

```
yaw = np.deg2rad(-90.0)
rotation_matrix = np.array([[np.cos(yaw), -np.sin(yaw), 0.0],
                            [np.sin(yaw), np.cos(yaw), 0.0], [0.0, 0.0, 1.0]])

rotated_lidar = (rotation_matrix @ (lidar).T).T
rotated_lidar[:, 1] *= -1
```

KP-Zhang, 2022-08-31:
Thank you for your reply. It is very helpful. Thanks.

closed 2022-08-31 by KP-Zhang

BestBUCKY, 2022-10-12:
> Depends on where you read the point cloud. I assume in the following that you directly take it from the tick_data['lidar']. The default coordinate system that CARLA uses is x front, y right, z up, [...]

hello， im confused about that “then use only the front frames by doing an action repeat of 2 for the model”，i didn’t find this action in the code，can you tell me how to do that？ thanks