autonomousvision/transfuser #102: cuda version

yanzhaohui1124, 2022-09-05:
can you tell your cuda version and nvidia version

Kait0, 2022-09-05:
The code base is set up to run with cuda version 10.2 (Newer once like 11.3 should also work if you change the installed torch, torchvision, mmdet, mmcv and torch-scatter versions accordingly).

yanzhaohui1124, 2022-09-06:
thank you,i cannot use cuda10.2,so i will try again

yanzhaohui1124, 2022-09-06:
now,i can use **./datagen.sh**
but it shows that
**========= Preparing RouteScenario_0 (repetition 0) =========
> Setting up the agent
Town10HD_Scenario10_route0_09_06_14_56_30
> Loading the world
> Running the route
Sparse Waypoints: 4
Dense Waypoints: 86
/home/tasl/transfuser/team_code_autopilot/data_agent.py:547: UserWarning: Creating a tensor from a list of numpy.ndarrays is extremely slow. Please consider converting the list to a single numpy.ndarray with numpy.array() before converting to a tensor. (Triggered internally at  /opt/conda/conda-bld/pytorch_1659484809535/work/torch/csrc/utils/tensor_new.cpp:201.)
  template_batched_torch = torch.tensor(template_batched, device='cuda', dtype=torch.float32).unsqueeze(1)
/home/tasl/anaconda3/envs/tfuse/lib/python3.7/site-packages/numpy/lib/npyio.py:528: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  arr = np.asanyarray(arr)
**
is this normal?

yanzhaohui1124, 2022-09-06:
I meet the error _UnboundLocalError: local variable 'leaderboard_evaluator' referenced before assignment
_,can you help me?

Kait0, 2022-09-06:
I think the performance warning won't matter much.
For the second error you need to provide the full stack trace.
The real issue usually comes before the "local variable 'leaderboard_evaluator' referenced before assignment".
This is just a generic error.

yanzhaohui1124, 2022-09-06:
thank you very much

closed 2022-09-06 by yanzhaohui1124