autonomousvision/transfuser #104: i cant generate data

shenzo-ai, 2022-09-08:
i run the carla server 
then 
i run ./leaderboard/scripts/datagen.sh <carla root> <working directory of this repo (*/transfuser/)>
   it does not work and im getting this error  ; 
Traceback (most recent call last):
  File "/home/shenzo/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 31, in <module>
    from leaderboard.scenarios.scenario_manager_local import ScenarioManager
  File "/home/shenzo/transfuser/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 25, in <module>
    from leaderboard.autoagents.agent_wrapper_local import AgentWrapper, AgentError
  File "/home/shenzo/transfuser/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 19, in <module>
    from leaderboard.autoagents.autonomous_agent import Track
  File "/home/shenzo/transfuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>
    from leaderboard.utils.route_manipulation import downsample_route
  File "/home/shenzo/transfuser/leaderboard/leaderboard/utils/route_manipulation.py", line 17, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents'

Kait0, 2022-09-08:
You need to add the CARLA folders to your python path:
path/to/CARLA/PythonAPI
/path/to/CARLA/PythonAPI/carla

shenzo-ai, 2022-09-08:
thank you so much, but now im having this error , would please help

Traceback (most recent call last):
  File "/home/shenzo/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 32, in <module>
    from leaderboard.scenarios.route_scenario_local import RouteScenario
  File "/home/shenzo/transfuser/leaderboard/leaderboard/scenarios/route_scenario_local.py", line 26, in <module>
    from srunner.scenarioconfigs.scenario_configuration import ScenarioConfiguration, ActorConfigurationData
  File "/home/shenzo/transfuser/scenario_runner/srunner/scenarioconfigs/scenario_configuration.py", line 65, in <module>
    class ScenarioConfiguration(object):
  File "/home/shenzo/transfuser/scenario_runner/srunner/scenarioconfigs/scenario_configuration.py", line 83, in ScenarioConfiguration
    weather = carla.WeatherParameters()
AttributeError: module 'carla' has no attribute 'WeatherParameters'

Kait0, 2022-09-08:
Have you added CARLA itself to the parth?
 /path/to/CARLA/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg

Kait0, 2022-09-08:
In principle all of these should be set automatically by the datagen.sh
Are you calling it with both parameters, the carla root and the transfuser root?
./leaderboard/scripts/datagen.sh <carla root> <working directory of this repo (*/transfuser/)>
In your description above it sounds like you forgot the carla root parameter

shenzo-ai, 2022-09-08:
im just typing  :  ./leaderboard/scripts/datagen.sh 
in the transfuser directory 
and after i added the carla to the path , im getting this error now (super thankful for your help ) :
 ./leaderboard/scripts/datagen.sh

shenzo-ai, 2022-09-08:
im just typing : ./leaderboard/scripts/datagen.sh
in the transfuser directory
and after i added the carla to the path , im getting this error now (super thankful for your help ) : 
Traceback (most recent call last):
  File "/home/shenzo/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 24, in <module>
    import carla
  File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/__init__.py", line 8, in <module>
  File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 9, in <module>
  File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in __bootstrap__
ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory

shenzo-ai, 2022-09-08:
it is fixed now , i just want to thank youuuuuuuuuuuuuuuuuuuuuuuu very much for you're time

closed 2022-09-08 by Kait0

MCUBE-2023, 2024-05-17:
Dear author, 
Although I made sure to set up the path correctly according to the roots of the repos on the server on which I am working by replacing the default paths in the first 2 lines for each of these files: datagen.sh as well as local_evaluation.sh (please see the 2 screenshots of how I modified the paths), when running ./leaderboard/scripts/local_evaluation.sh, I got the same issue as @shenzo-ai which is: Traceback (most recent call last):
File "/home/shenzo/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 24, in
import carla
File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/init.py", line 8, in
File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 9, in
File "/home/shenzo/transfuser/carla/PythonAPI/carla/dist/carla-0.9.10-py3.7-linux-x86_64.egg/carla/libcarla.py", line 7, in bootstrap
ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory

[image]
[image]

In addition, when I run:  ./leaderboard/scripts/local_evaluation.sh <carla root> <working directory of this repo (*/transfuser/)> and I define explicitly the paths as arguments for both CARLA_ROOT and WORK_DIR which gives me this instruction: ./leaderboard/scripts/local_evaluation.sh /SHARED/marwen/transfuser/carla//SHARED/marwen/transfuser, I got this error: 

Traceback (most recent call last):
  File "/SHARED/marwen/transfuser/leaderboard/leaderboard/leaderboard_evaluator_local.py", line 31, in <module>
    from leaderboard.scenarios.scenario_manager_local import ScenarioManager
  File "/SHARED/marwen/transfuser/leaderboard/leaderboard/scenarios/scenario_manager_local.py", line 25, in <module>
    from leaderboard.autoagents.agent_wrapper_local import AgentWrapper, AgentError
  File "/SHARED/marwen/transfuser/leaderboard/leaderboard/autoagents/agent_wrapper_local.py", line 19, in <module>
    from leaderboard.autoagents.autonomous_agent import Track
  File "/SHARED/marwen/transfuser/leaderboard/leaderboard/autoagents/autonomous_agent.py", line 17, in <module>
    from leaderboard.utils.route_manipulation import downsample_route
  File "/SHARED/marwen/transfuser/leaderboard/leaderboard/utils/route_manipulation.py", line 16, in <module>
    from agents.navigation.global_route_planner import GlobalRoutePlanner
ModuleNotFoundError: No module named 'agents'

Would you please help me in resolving these errors ? Thanks!

Kait0, 2024-05-18:
>  ImportError: libjpeg.so.8: cannot open shared object file: No such file or directory

This problem happens when libjpeg file is not installed in your operating system.
You can either install it or upload the file to another folder and link it.
What I usually do is link the miniconda lib folder which often contains the missing lib files.
See [here](https://github.com/autonomousvision/carla_garage/blob/ae7e83f0a2e33e094cd1b369f51677c7467d9eb2/evaluate_routes_slurm.py#L19) for example.

> /SHARED/marwen/transfuser/carla//SHARED/marwen/transfuser

Maybe the double // after CARLA is a problem?

MCUBE-2023, 2024-05-21:
Thank you so much for your quick reply! The paths that I defined were set correctly. However, the problem was indeed in installing libjpeg. The issue is resolved and thanks again!