autonomousvision/transfuser #106: Data generation at 2 fps

cozeybozey, 2022-09-13:
I saw in the paper that it says you are collecting data at 2 fps and I was wondering what that meant exactly. Do you let the simulation run at higher fps, let's say for example 20 fps and then only collect data for 2 frames each second? Or is everything just running at 2 fps including the simulation itself? I can imagine you would do the former to avoid having many frames that are very similar to each other. What is your opinion on this? If it is the latter, then why would you set it at 2 fps? Is that just the fastest you can get?

Kait0, 2022-09-13:
The frames per second here refer to seconds in simulation time not real time (one simulator step in CARLA is always 50 ms simulation time but real time can change with hardware etc).
We let the simulation run at the higest possible frame rate (in real time) of course to collect data as fast as possible.
Collecting data at 2 FPS means we only store data every 10th step of the simulator.
The purpose is to reduce the amount of data that is very similar to each other.

cozeybozey, 2022-09-13:
I see that is indeed what I suspected, thanks for the reply!

closed 2022-09-13 by cozeybozey