autonomousvision/transfuser #108: Confusion about inverting x,y

Abyss-J, 2022-09-30:
Thanks for your great works! It inspires me a lot. And I met some problems that bothered me.
As I know, the x, y and z axes in Carla point to the front, right and top respectively. When converting the coordinate system in your code, set `seq_ x[i]` and `seq_ y[i]` takes the opposite number, that is, it reverses the x and y axes, but the `full_lidar[i]` only seems to inverts the y axis, and the comment is that it reverses the x and y axes. I am puzzled about this
```
# process only past lidar point clouds
            if i < self.seq_len:
                # convert coordinate frame of point cloud
                full_lidar[i][:,1] *= -1 # inverts x, y
                full_lidar[i] = transform_2d_points(full_lidar[i], 
                    np.pi/2-seq_theta[i], -seq_x[i], -seq_y[i], np.pi/2-ego_theta, -ego_x, -ego_y)
                lidar_processed = lidar_to_histogram_features(full_lidar[i], crop=self.input_resolution)
                data['lidars'].append(lidar_processed)
```

Kait0, 2022-10-01:
The comment is incorrect it only inverts one axis of course.
If you want to understand the coordinate system of the LiDAR perhaps this is helpful:
https://github.com/autonomousvision/transfuser/issues/100#issuecomment-1232591493

Abyss-J, 2022-10-02:
Thank you for your reply. It is very helpful. Thanks.

closed 2022-10-02 by Abyss-J