autonomousvision/transfuser #109: about part of transformer in transfuser model

t19cs008, 2022-10-03:
I have a question about part of transformer.

https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/config.py#L126-L129

https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/transfuser.py#L19-L20

https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/transfuser.py#L34-L46

In this this config, parameter is defined and applied to AdaptiveAvgPool layer setting and transformer setting.

My question is bellow comment.

https://github.com/autonomousvision/transfuser/blob/bef1fcf5622c3ab400d4c1f27059786e7cfecb11/team_code_transfuser/transfuser.py#L146-L149

I think, the output shape is fixed when using AdaptiveAvgPool, so in this case image fusion shape would be (B, 72, 5, 22) and lidar fusion shape would be (B, 72, 8, 8).

Please tell me why image fusion shape will be (B, 72, 40, 176) and lidar fusion shape would be (B, 72, 64, 64).

Kait0, 2022-10-03:
The comment is just referencing the image resolution before the avgpooling.
You are correct the sensor fusion always happens at the downscaled resolution.

t19cs008, 2022-10-04:
Thank you for your reply!
My question is solved.

closed 2022-10-04 by t19cs008