autonomousvision/transfuser #112: some problems about lidar data

BestBUCKY, 2022-10-11:
hello，I have some questions about lidar data that i would like to ask you

1. What is the coordinate system of lidar？ Is that the left is the positive direction of the x-axis, the front is the positive direction of the y-axis, and the lower is the positive direction of the z-axis？

2. I visualized the lidar data, it looks like the right and left are reversed. Is it right？
[image]
[image]

3.In your lidar data process code. It seems like only invert y-axis? And it is already in the local coordinate system, why still need to convert the coordinate system？
            # process only past lidar point clouds
            if i < self.seq_len:
                # convert coordinate frame of point cloud
                full_lidar[i][:,1] *= -1 # inverts x, y
                full_lidar[i] = transform_2d_points(full_lidar[i], 
                    np.pi/2-seq_theta[i], -seq_x[i], -seq_y[i], np.pi/2-ego_theta, -ego_x, -ego_y)
                lidar_processed = lidar_to_histogram_features(full_lidar[i], crop=self.input_resolution)
                data['lidars'].append(lidar_processed)

I would be very grateful if you could answer me.

ap229997, 2022-10-11:
Check https://github.com/autonomousvision/transfuser/issues/100#issuecomment-1232591493, it should be helpful.

BestBUCKY, 2022-10-11:
> Check [#100 (comment)](https://github.com/autonomousvision/transfuser/issues/100#issuecomment-1232591493), it should be helpful.

Thanks! it helps me a lot

closed 2022-10-11 by BestBUCKY